embassy-rs/embassy #1841: Multiple USB endpoints with same driver

0xa10, 2023-08-29:
Hi,
I'm toying with a HID device and I'd like to also log to a serial port. 
All the current examples I've seen only implement one USB endpoint.

Looking at `UsbDevice::build`  - it consumes the Driver object when calling `start` on the driver:
https://github.com/embassy-rs/embassy/blob/main/embassy-usb/src/lib.rs#L233
https://github.com/embassy-rs/embassy/blob/main/embassy-usb-driver/src/lib.rs#L166

The reasoning seems to be that all endpoint allocations should occur before the peripheral itself is started.
But unless I'm missing something there is no pattern that allows multiple endpoints to be allocated before `start` is invoked - perhaps the call to `start` should be abstracted out from the UsbDevice build function?

Dirbaio, 2023-08-29:
You can create a composite device by creating multiple classes on the builder:

```rust
    let mut builder = Builder::new(...);
    let serial_class = CdcAcmClass::new(&mut builder, ...);
    let mut hid_writer = HidWriter::<_, 5>::new(&mut builder, &mut state, config);
    let mut usb = builder.build();
    ...

0xa10, 2023-09-30:
Whoops, forgot about opening this issue - thanks, I understand what I missed now 👍

closed 2023-09-30 by 0xa10