embassy-rs/embassy #1853: example: std net `ConnectionReset` on Ubuntu 22.04

LimpidCrypto, 2023-09-01:
I'm currently trying to implement [embedded-io-async](https://github.com/rust-embedded/embedded-hal/tree/master/embedded-io-async) for [embedded-async](https://github.com/ninjasource/embedded-websocket) to use the websocket with `embassy-net`.

I have followed the instructions in the [README](https://github.com/embassy-rs/embassy/blob/main/examples/std/README.md) file to set up `tap0`:
```bash
tap0: flags=4099<UP,BROADCAST,MULTICAST>  mtu 1500
        inet 192.168.69.100  netmask 255.255.255.0  broadcast 0.0.0.0
        inet6 fe80::f884:56ff:fe3b:b1e1  prefixlen 64  scopeid 0x20<link>
        ether fa:84:56:3b:b1:e1  txqueuelen 1000  (Ethernet)
        RX packets 273  bytes 13622 (13.6 KB)
        RX errors 0  dropped 0  overruns 0  frame 0
        TX packets 81  bytes 7858 (7.8 KB)
        TX errors 0  dropped 0 overruns 0  carrier 0  collisions 0
```
Then I had netcat listen to port `8000` using `nc -l 8000`. I just copied your current std net example and ran it like so: 
`cargo run --example client_async_embedded_io_async --features "embedded-io-async" -- --static-ip`

I'm getting a `ConnectionReset` error. Are there common sources for the error?
```bash
[2023-09-01T18:54:15.110579234Z DEBUG embassy_net] IPv4: UP
[2023-09-01T18:54:15.110643195Z DEBUG embassy_net]    IP address:      Cidr { address: Address([192, 168, 69, 2]), prefix_len: 24 }
[2023-09-01T18:54:15.110659886Z DEBUG embassy_net]    Default gateway: Some(Address([192, 168, 69, 1]))
[2023-09-01T18:54:15.110671398Z INFO  embassy_net] IPv6: DOWN
[2023-09-01T18:54:15.110699561Z INFO  client_async_embedded_io_async] connecting to (Address([192, 168, 69, 100]), 8000)...
[2023-09-01T18:54:15.110789350Z INFO  embassy_net] link_up = true
[2023-09-01T18:54:25.110887049Z WARN  client_async_embedded_io_async] connect error: ConnectionReset
```

Dirbaio, 2023-09-01:
are you using GNU netcat or BSD netcat? `nc -l 8000` works for BSD netcat, for GNU the flags are different (not sure which).

jubeormk1, 2025-05-21:
Hi there,

I am also troubleshooting embassy net on an ubuntu machine (fresh ubuntu 24.04) and the example is waiting for you to listen to a connection in the port 8000. The hint given by @Dirbaio works.

I have to say though that I am using the [current main branch](https://github.com/embassy-rs/embassy/commit/7cbc9058bc726900571a7858c581f60cd8cb0266). Following the instructions plus listening on the correct port (`nc -l 8000`) makes net to send a cascade of Hello!s.

I recommend closing this issue.

closed 2025-05-21 by Dirbaio