embassy-rs/embassy #1866: Hardware Compatible

chen-gz, 2023-09-05:
Hi,

I am trying to use the embassy for my embedded development. I have found that Rust async is really nice for IO functions. However, when I attempted to use the examples for my hardware, I discovered that they were not working correctly. I had to make some changes to the code in the HAL library.

Could you provide a list of the hardware that has been tested with each example?

Thank you.

xoviat, 2023-09-06:
It's not a surprise since we are attempting to support many targets. All you can do is tell us whether something is not working on a specific target, and we can try to fix it.

chen-gz, 2023-09-06:
I am using (NUCLEO-U575ZI-Q)[https://www.st.com/en/evaluation-tools/nucleo-u575zi-q.html] to test the library. I first tested the simple I2C. The clock is not set up to the correct frequency. Then I also test use_serial, it also does not work. Also, I tested the RTC, There are some errors in the RTC setup. I tested SDMMC on a custom board, and it stuck in the initialization process. I know my hardware has some issues, but for a rust code, it should give an error instead of blocking it.

xoviat, 2023-09-06:
stm32u585ai is the closest board we have that's treated. Rtc is not yet tested on that target, but i2c should work.

chen-gz, 2023-09-06:
I2C is working, but the frequency is not setup correctly.
For example, I put 400Khz when initialization i2c instance. But I will get about 100Khz instead. The device will response and working fine. But the frequency actually is not correct.
I don't remember what exactly frequency I got. But they are not match.  This can be observed from a scope.

The problem of RTC is the RCC register not setup correct. If I got time I can figure out how to setup clock under embassy framework. I just modify the RCC register to make it work for now.

xoviat, 2023-09-06:
See #1868 for an example of what needs to be done.

closed 2025-01-10 by Dirbaio