embassy-rs/embassy #1876: new_output_a not working correctly rp2040

ProjectInitiative, 2023-09-08:
```Rust
//! This example shows how to use PWM (Pulse Width Modulation) in the RP2040 chip.
//!
//! The LED on the RP Pico W board is connected differently. Add a LED and resistor to another pin.

#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt::*;
use embassy_executor::Spawner;
use embassy_rp::pwm::{Config, Pwm};
use embassy_time::{Duration, Timer};
use {defmt_rtt as _, panic_probe as _};

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let p = embassy_rp::init(Default::default());

    let mut c: Config = Default::default();
    // c.top = 0x1388;
    c.top = 5000;
    c.compare_b = c.top / 2;

    // let mut pwm = Pwm::new_output_a(p.PWM_CH4, p.PIN_8, c.clone());
    // let mut pwm = Pwm::new_output_b(p.PWM_CH4, p.PIN_9, c.clone());
    let mut pwm = Pwm::new_output_ab(p.PWM_CH4, p.PIN_8, p.PIN_9, c.clone());
    // let mut pwm2 = Pwm::new_output_ab(p.PWM_CH5, p.PIN_10, p.PIN_27, c.clone());
    pwm.set_config(&c);
    // pwm2.set_config(&c);

    loop {}
}
```

I have tried uncommenting every combination of the block about `loop {}`. I cannot get any output on A for pins 8 or 10, or others that I have tried. I have swapped channels on my oscope for a sanity check. This is what I am expecting:

[image]

Vs what I get:

[image]

Dirbaio, 2023-09-08:
have you tried `c.compare_a = c.top / 2;` ?

ProjectInitiative, 2023-09-08:
I have been staring at code too long. That is the issue.

closed 2023-09-08 by ProjectInitiative