embassy-rs/embassy #1879: DMA ringbuffer needs a read_half/write_half async function that uses the half complete interrupt

tyler-gilbert, 2023-09-10:
Streaming data use cases (reading audio on ADC, writing audio to I2S, and more) can leverage the half-complete and complete interrupts to allow a loop to read/write data using a task that await's the half complete and complete interrupts.

Here is the pseudo-code:

```
//First fill the full buffer by writing two halves
i2s.write_half(buffer).await;
i2s.write_half(buffer).await;
i2s.start(); //starts up as a circular DMA buffer whose size is 2*buffer.len()

loop {
   wav_file.read(buffer); // for example, reads audio samples from flash memory
  // 1st iteration: this should await until the half complete interrupt wakes
  // 2nd iteration: this should await until the transfer complete interrupt wakes
  // ... : every other iteration will write to the inactive half of the buffer and await the next half/complete interrupt
   i2s.write_half(buffer).await; 
}
```

This could use a new `dma::ringbuffer` (maybe `dma::half_ring_buffer`) interface. It would need another entry in the `struct State` to keep track of `half_complete_count`. Rather than tracking the DMA register for words transferred, there are always either 0 or N/2 words available to read. Whether the first half or second half is currently readable is based on the interrupt.

I can work on a PR. I just need some help to know if this is the right approach.

xoviat, 2023-09-10:
What is the advantage of this over read/write exact?

alexferro, 2023-09-10:
Yeah, Tyler, `read_exact` is the exact feature I added to embassy for this usecase, and I've been using in production. I don't think this is relevant. Is it no longer working?

tyler-gilbert, 2023-09-10:
@alexferro and @xoviat,

I see now. `read_exact` will work. The logic is quite a bit simpler if we assume we are always reading `N/2` from the ring buffer. But probably not worth the effort.

I will close this one.

closed 2023-09-10 by tyler-gilbert