embassy-rs/embassy #1900: Panicking during TaskStorage initialization leaks TaskStorage

bugadani, 2023-09-14:
After unwinding, the TaskStorage is left in `SPAWNED | RUN_QUEUED` state without actually being registered to the executor, which means the storage can't be reused later.

cc https://github.com/embassy-rs/embassy/pull/1897 which has a solution that is not zero-cost.

jamesmunns, 2023-11-28:
Just driving by, is this particularly for cases where `panic = "unwind"`? I'm not aware of any embedded platforms where this is the case (does esp-idf support unwinding?), is this only for desktop uses of `embassy`?

In the case of `panic = "abort"`, no unwinding (and no drop destructors) will be run at the time of panic, as the panic function is expected to diverge.

jamesmunns, 2023-11-28:
Update: Nevermind, dirbaio outlined this in https://github.com/embassy-rs/embassy/pull/1897#issuecomment-1721649449. Apologies for the noise :)