embassy-rs/embassy #1904: nRF52 Blinky: No loadable ELF sections were found

mbardwell, 2023-09-14:
# Summary
I want to compile blinky for my nRF52 board. It works if I run the nRF52840 example (with nRF52832 modifications) from this repository, but I would like to set it up as a stand-alone project. It compiles, but I get this error when I try to flash the board.

```
$ cargo +nightly run --release --example blinky
...
    Finished release [optimized] target(s) in 21.03s
     Running `probe-rs run --chip nRF52832_xxAA 'target\thumbv7em-none-eabi\release\examples\blinky'`
Error: No loadable ELF sections were found.
error: process didn't exit successfully: `probe-rs run --chip nRF52832_xxAA 'target\thumbv7em-none-eabi\release\examples\blinky'` (exit code: 1)
```

# Setup
Hardware: nRF52832 DK plugged into Windows 11 PC

Directory structure. All non-generated files are copied from the nRF52840 example (with nRF52832 modifications), with the exception of Cargo.toml, which I have included below.
```
blinky/
├───.cargo/cargo.toml
├───examples/blinky.rs
├───src/main.rs
└───target
    └───release
    └───thumbv7em-none-eabi
        └───release
            ├───examples/blinky  <-- generated ELF binary
            └───incremental
├───build.rs
├───Cargo.toml
├───Cargo.lock
├───memory.x
```

Cargo.toml
```
[package]
name = "blinky"
version = "0.1.0"
edition = "2021"

[features]
default = ["nightly"]
nightly = [
    "embassy-executor/nightly",
    "embassy-nrf/nightly",
    "embassy-time/nightly",
    "embassy-time/unstable-traits",
]

[dependencies]
embassy-executor = { version = "0.3.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "nightly", "integrated-timers", "arch-cortex-m", "executor-thread"] }
embassy-time = { version = "0.1.3", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "nightly"] }
embassy-nrf = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "nrf52832", "time-driver-rtc1", "gpiote", "nightly"] }

defmt = "0.3"
defmt-rtt = "0.3"

cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
cortex-m-rt = "0.7.0"
panic-probe = { version = "0.3", features = ["print-defmt"] }
```

Here is a preview of the generated ELF that doesn't load
* Binary is 967112 bytes. The fact this this is 1MB smaller than the working binary tells me something is wrong.
```
00000000   7F 45 4C 46 01 01 01 00 00 00 00 00 00 00 00 00  ELF............
00000010   02 00 28 00 01 00 00 00 00 00 00 00 34 00 00 00  ..(.........4...
00000020   48 BF 0E 00 00 02 00 05 34 00 20 00 03 00 28 00  H¿......4. ...(.
00000030   10 00 0E 00 06 00 00 00 34 00 00 00 34 00 01 00  ........4...4...
00000040   34 00 01 00 60 00 00 00 60 00 00 00 04 00 00 00  4...`...`.......
00000050   04 00 00 00 01 00 00 00 00 00 00 00 00 00 01 00  ................
...
```

Here is a preview of the generated ELF that does load
* Binary is 1940896 bytes
```
00000000   7F 45 4C 46 01 01 01 00 00 00 00 00 00 00 00 00  ELF............
00000010   02 00 28 00 01 00 00 00 DD 00 00 00 34 00 00 00  ..(.....Y...4...
00000020   E0 99 1D 00 00 02 00 05 34 00 20 00 06 00 28 00  à?......4. ...(.
00000030   18 00 16 00 01 00 00 00 14 01 00 00 00 00 00 00  ................
00000040   00 00 00 00 DC 00 00 00 DC 00 00 00 04 00 00 00  ....Ü...Ü.......
00000050   04 00 00 00 01 00 00 00 F0 01 00 00 DC 00 00 00  ........d...Ü...
00000060   DC 00 00 00 4C 30 00 00 4C 30 00 00 05 00 00 00  Ü...L0..L0......
00000070   04 00 00 00 01 00 00 00 3C 32 00 00 28 31 00 00  ........<2..(1..
00000080   28 31 00 00 BC 04 00 00 BC 04 00 00 04 00 00 00  (1..¼...¼.......
00000090   04 00 00 00 01 00 00 00 F8 36 00 00 00 00 00 20  ........o6.....
000000A0   E8 35 00 00 70 00 00 00 70 00 00 00 06 00 00 00  è5..p...p.......
```

Happy to provide more information to assist with the debug.

Dirbaio, 2023-09-14:
most common cause for empty ELFs is you're missing these lines in build.rs https://github.com/embassy-rs/embassy/blob/main/examples/nrf52840/build.rs#L32-L34

if that still doesn't fix it please post the entire project as a github repo.

mbardwell, 2023-09-15:
I have those lines in my build.rs. I pushed the project [to its own repository](https://github.com/mbardwell/rust-nrf-blinky).

Dirbaio, 2023-09-15:
`cargo:rustc-link-arg-bins` only applies to binaries (`src/main.rs`, `src/bin/foo.rs`), not to examples (`examples/foo.rs`)

you can make your code a bin instead of an example, or you can duplicate the lines with `cargo:rustc-link-arg-examples`. see [docs](https://doc.rust-lang.org/cargo/reference/build-scripts.html#outputs-of-the-build-script).

mbardwell, 2023-09-15:
Adding these lines to build.rs did the trick
```
    println!("cargo:rustc-link-arg-examples=--nmagic");
    println!("cargo:rustc-link-arg-examples=-Tlink.x");
    println!("cargo:rustc-link-arg-examples=-Tdefmt.x");
```

Full build.rs
```
//! This build script copies the `memory.x` file from the crate root into
//! a directory where the linker can always find it at build time.
//! For many projects this is optional, as the linker always searches the
//! project root directory -- wherever `Cargo.toml` is. However, if you
//! are using a workspace or have a more complicated build setup, this
//! build script becomes required. Additionally, by requesting that
//! Cargo re-run the build script whenever `memory.x` is changed,
//! updating `memory.x` ensures a rebuild of the application with the
//! new memory settings.

use std::env;
use std::fs::File;
use std::io::Write;
use std::path::PathBuf;

fn main() {
    // Put `memory.x` in our output directory and ensure it's
    // on the linker search path.
    let out = &PathBuf::from(env::var_os("OUT_DIR").unwrap());
    File::create(out.join("memory.x"))
        .unwrap()
        .write_all(include_bytes!("memory.x"))
        .unwrap();
    println!("cargo:rustc-link-search={}", out.display());

    // By default, Cargo will re-run a build script whenever
    // any file in the project changes. By specifying `memory.x`
    // here, we ensure the build script is only re-run when
    // `memory.x` is changed.
    println!("cargo:rerun-if-changed=memory.x");

    println!("cargo:rustc-link-arg-bins=--nmagic");
    println!("cargo:rustc-link-arg-bins=-Tlink.x");
    println!("cargo:rustc-link-arg-bins=-Tdefmt.x");

    println!("cargo:rustc-link-arg-examples=--nmagic");
    println!("cargo:rustc-link-arg-examples=-Tlink.x");
    println!("cargo:rustc-link-arg-examples=-Tdefmt.x");
}
```

Thanks for your help @Dirbaio !

closed 2023-09-15 by mbardwell