embassy-rs/embassy #1905: Why not published on crates.io?

SploxFox, 2023-09-14:
Hello, why are most of the embassy crates not yet published on Crates.io?

xoviat, 2023-09-15:
Crates are published as-able. We've found that primarily using git versions, along with a CI and HIL (hardware-in-loop) pipeline allows rapid iteration and refactoring as-needed across the multiple crates that comprise the project.

lulf, 2023-09-15:
@SploxFox For a create to be released, we require that all public APIs should be documented and no big refactorings needed before releasing.

You can track the status of different crates here (may be slightly out of date): https://github.com/embassy-rs/embassy/issues/1050

justin-lovell, 2023-10-01:
To be fair, there are numerous projects which don't publish a detailed documentation before publishing to `crates.io`.

The biggest problem created from the perfectionism is that it makes Embassy unapproachable. Does a developer use a git reference in their Cargo file to a tag? No, half of the packages are published as tags. Would a developer do a git submodule link? Perhaps.

Just a simple document with a guidance on how developers should interact with Embassy when not using the standard publishing method goes a long way.

To say a project to have batteries included, it needs to have a power switch when the documentation does not disclose the real state.

closed 2023-12-08 by Dirbaio