embassy-rs/embassy #1922: Waiting on Signal from multiple tasks causes infinite wakeups

haileys, 2023-09-18:
Reproducer:

```rust
static SIGNAL: Signal<CriticalSectionRawMutex, ()> = Signal::new();

#[embassy_executor::task]
async fn one() {
    let mut wait_fut = SIGNAL.wait();

    poll_fn(|cx| {
        log::info!("polling one");
        wait_fut.poll_unpin(cx)
    }).await;
}

#[embassy_executor::task]
async fn two() {
    let mut wait_fut = SIGNAL.wait();

    poll_fn(|cx| {
        log::info!("polling two");
        wait_fut.poll_unpin(cx)
    }).await;
}
```

Log output:

```
INFO - polling two
INFO - polling one
INFO - polling two
INFO - polling one
INFO - polling two
INFO - polling one
INFO - polling two
INFO - polling one
INFO - polling two
INFO - polling one
INFO - polling two
INFO - polling one
INFO - polling two
... and so on and so forth ...
```

This happens because `Signal::poll_wait` only has one waker slot. There is a branch which coalesces similar wakers together (by checking if one will wake the other with `will_wake`), but if they are two different wakers, the most recent waker will steal the slot and wake the other. This causes the loop.

I think to fix this in general, we either need dynamic allocation so that we can keep track of a dynamic list of waiters, or we need something like `MultiWakerRegistration` which uses a fixed size vec. Neither of these solutions are optimal though, dynamic allocation for obvious reasons (it can't be taken as a given in embedded contexts), and `MultiWakerRegistration` because the const `N` parameter would need to be threaded back through the `Signal` API.

One potential solution could be to use a bitset to encode which tasks are waiting on a future. Because all tasks are statically allocated up front, we know exactly how many bits of state we need to keep. I think it's a reasonable assumption in most cases that the number of statically allocated tasks will be less than 64 (the number of bits required to store a non-turbo `Waker`), but some kind of escape hatch could be provided for cases where there are more tasks. This only works if no other libraries create wakers, but this is the same constraint as in `task_from_waker` and the turbo wakers optimisation.

lulf, 2023-09-18:
According to the embassy-sync docs, Signal is a "Signalling latest value to a single consumer." but maybe it can be better documented on the type itself. The general philosophy I think is that the overhead of Signal should be as small as possible.

If you want broadcast-like semantics, maybe you could use https://docs.rs/embassy-sync/0.3.0/embassy_sync/pubsub/struct.PubSubChannel.html# ?

There is also Channel which allows two tasks to compete for a value.

haileys, 2023-09-18:
> According to the embassy-sync docs, Signal is a "Signalling latest value to a single consumer."

This is true, but this requirement is not enforced by the type system. It would be easy to misuse a Signal, especially as a codebase evolves with multiple contributors. I think it would be reasonable to make Signal panic here, and even better if this restriction could somehow be checked at compile time, but the current behaviour in this situation does need to be fixed somehow.

tact1m4n3, 2024-05-24:
I think a good solution to this is to create a `Sender` and a `Receiver` struct for `Signal` which would allow only a single instance of either of these structs to be created. Interacting with the signal would happen through the sender/receiver. Much like the other channel types.