embassy-rs/embassy #1954: Support for input capture modes on STM32

dberlin, 2023-09-26:
Hey folks,
It looks like embassy-stm32 does not support an easy interface to the input capture mode, only the output modes.

If nobody is working on this, i'll add support to CaptureCompare{16,32}bitInstance and friends for the input capture stuff, and track it here.

I assume it belongs in this trait, rather than creating a new input trait.

I don't want to try to design a high level timer input capture API ATM, just get the medium-level abstraction to be nicer, so that you can easily set input channel polarities and whatnot.

xoviat, 2023-09-27:
Aren't those sealed traits?

dberlin, 2023-09-28:
They are sealed, but not inaccessible.  They are exposed through the low_level timer API.    So you can't make new impls of them, but you can use them and call their methods.
See examples/stm32h7/src/bin/low_level_timer_api.rs for one example usage.

You can use the traits through  "use embassy_stm32::timer::low_level::CaptureCompare32bitInstance" and then call the methods.  You just implement the trait yourself.

xoviat, 2023-09-28:
@dberlin I've merged your prs, but to be clear, the project leader has decided that the low-level traits are deprecated and will be removed when a sufficient high-level abstraction becomes available.

dberlin, 2023-09-28:
First of all, thanks.
Second, i think that's a reasonable position.  You'd need this to do a reasonable job of that anyway.
My use case is actually simple - i'm performing fast pulse width measurement for a weirdly modulated RF protocol.
So i use timer input capture set up to capture both the rising and falling edges of the same pin.
The only slightly tricky thing I do is that the STM32 has a mode to measure the same pin on two different input channels (so you can tell whether it was a rising/falling edge without reading the pin explicitly)

Most input capture abstractions you can think of would work fine for this.

I'm actually happy to add a simple PulseWidthMeasurement abstraction if anyone is interested

xoviat, 2023-09-29:
Yes, if you can add something like the Qei or Pwm abstraction (but for capture), that would be welcome.

brunob45, 2024-05-06:
Anyone is working on this subject? I also have a project where I want to acurately measure pusle width, and ideally using async.

I started toying with a `InputCapture` based on `SimplePWM`, and it is working pretty well for sync operations (#2912).

 Now I'll try to add interrupt capabilities inspired from other's peripherals `State`, like `I2C` or `USART`.

romainreignier, 2024-05-23:
I am porting an old C project in which I have used a Timer in PWM Input mode in order to measure a pulse width (for the common HC-SR04 ultrasonic sensor).
I see that for now, the `PWM Input mode` is not implemented with a high-level interface.
How do you measure a pulse width with only the Input Capture interface? By reading the state and if it is low, reading the capture value?

brunob45, 2024-05-26:
@romainreignier, since PR #2912 was merged, I added basic support for input capture, as seen in [this example](https://github.com/embassy-rs/embassy/blob/main/examples/stm32f4/src/bin/input_capture.rs).

With this, you can call `ic.wait_for_rising_edge(Channel::Ch3).await` to get the timer value when a rising edge is detected. Call this 2 times in a row, compute the difference, and you have the period duration of the pulse. You can set the `tick` duration in the `input_capture::new` constructor function.

Otherwise, would you have any suggestion to make this feature easier to use? What looks like the C code of your old project?

romainreignier, 2024-05-27:
Hi @brunob45 Thanks for your comment.

> since PR #2912 was merged, I added basic support for input capture, as seen in this example.

I have seen your PR but I did not managed to get the example working on a stm32f103 and stm32f411 boards.

Output:

```
0.000000 INFO  Hello World!
└─ input_capture::____embassy_main_task::{async_fn#0} @ src/bin/input_capture.rs:38  
0.000091 INFO  wait for risign edge
└─ input_capture::____embassy_main_task::{async_fn#0} @ src/bin/input_capture.rs:46  
0.000213 INFO  high
└─ input_capture::__blinky_task::{async_fn#0} @ src/bin/input_capture.rs:21  
0.300292 INFO  low
└─ input_capture::__blinky_task::{async_fn#0} @ src/bin/input_capture.rs:25  
0.600402 INFO  high
```

The `wait_for_rising_edge()` never returns while replacing the main of the example with the following to test the hardware setup (use input interrupt and `Instant` to compute period with software) works as expected:

```rust
#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let p = embassy_stm32::init(Default::default());
    info!("Hello World!");

    unwrap!(spawner.spawn(blinky(p.PB2)));

    let mut button = ExtiInput::new(p.PB10, p.EXTI10, Pull::None);

    loop {
        let t = Instant::now();
        info!("wait for rising edge");
        button.wait_for_rising_edge().await;
        info!("new period: {} ms", t.elapsed().as_millis());
    }
}
```

Output:

```
0.600708 INFO  wait for rising edge
└─ input_capture::____embassy_main_task::{async_fn#0} @ src/bin/input_capture.rs:70  
0.900543 INFO  low
└─ input_capture::__blinky_task::{async_fn#0} @ src/bin/input_capture.rs:27  
1.200653 INFO  high
└─ input_capture::__blinky_task::{async_fn#0} @ src/bin/input_capture.rs:23  
1.200805 INFO  new period: 600 ms
```

> What looks like the C code of your old project?

As said, on my previous project, I was using the **PWM Input mode** of the timer which automatically stores the period in the `CCR1` register and the width in the `CCR2` register. In fact, I was using the ChibiOS HAL (see implementation [here](https://github.com/ChibiOS/ChibiOS/blob/203285ba8a60fe9bf2711fc121573db385d5b277/os/hal/ports/STM32/LLD/TIMv1/hal_icu_lld.h#L778) and [here](https://github.com/ChibiOS/ChibiOS/blob/203285ba8a60fe9bf2711fc121573db385d5b277/os/hal/ports/STM32/LLD/TIMv1/hal_icu_lld.c#L821-L822)).

Extract of the Reference Manual about the PWM Input mode:
[image]

I have checked `stm32-rs` project and they have a pwm_input implementation for [F1](https://github.com/stm32-rs/stm32f1xx-hal/blob/master/src/timer/pwm_input.rs) and [F4](https://github.com/stm32-rs/stm32f4xx-hal/blob/master/src/timer/pwm_input.rs).

I think we can keep the InputCapture module of embassy as is and add another high level PwmInput module so the user can choose the one that best fit its needs.

I might not be able to provide an implementation for now because I miss time and knowledge in low-level embedded rust dev.

brunob45, 2024-05-29:
I started to cook something up in PR #3014, if you want to look it up

romainreignier, 2024-05-29:
Thanks a lot for your work @brunob45

Dirbaio, 2025-01-10:
fixed in https://github.com/embassy-rs/embassy/pull/2912 https://github.com/embassy-rs/embassy/pull/3014 . for more low-level uses you can use the low-level timer driver.

closed 2025-01-10 by Dirbaio