embassy-rs/embassy #1961: Performance metrics – CPU time running/waiting?

Jajcus, 2023-09-28:
I am building a simple audio-processing device based on Rasbperry Pi Pico. Coded in Rust with Embassy, of course.

I would like to know how much performance margin I have. That would be easy with simple blocking main loop approach – I would just measure time left after each cycle, average it and report once in a while. The same is not trivial in asynchronous application with many different tasks running.

Is there a way to get similar metrics from Embassy executor – how much CPU time was spent running and how much waiting?

xoviat, 2023-09-28:
This can be accomplished by implementing your own executor and recording times before and after the raw executor is polled. In addition, you would need to record time spent in irq handlers to obtain a more accurate result.

Jajcus, 2023-09-29:
Thank you. This seems to be working for me. And good-enough without measuring the IRQ handlers (I had no idea how to do that anyway).

That is how I did it:
https://gist.github.com/Jajcus/d3407cd961b177d18f47477596e65108

The only think I don't like about it is that I had to redefine the magic ``THREAD_PENDER`` constant.