embassy-rs/embassy #1979: Infinite spin loop at initializing Timer in SyncExecutor (before entry funtion)

Dicklessgreat, 2023-09-30:
I am making a simple firmware that connects a sensor to the GPIO of a Raspberry Pi Pico W and sends its data via TCP to a server.
The spin loop is in the result of expanding the `#[embassy_executor::main]` macro before entering the entry function.
I use the latest embassy cloned from Github.
The call stack looks like this.

<details>
<summary>Call Stack</summary>

    1. __cpsie #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\asm\inline.rs:52)
    2. enable #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\call_asm.rs:11)
    3. <unknown function @ 0x10018284> #[inline] (c:\Users\Me\develop\MyFirmware\embassy\embassy-rp\src\critical_section_impl.rs:66)
    4. acquire #[inline] (c:\Users\Me\develop\MyFirmware\embassy\embassy-rp\src\critical_section_impl.rs:28)
    5. _critical_section_1_0_acquire #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src\lib.rs:282)
    6. acquire #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src\lib.rs:183)
    7. with<core::result::Result<u8, u8>, atomic_polyfill::polyfill::{impl#21}::fetch_update::{closure_env#0}<embassy_rp::timer::{impl#1}::allocate_alarm::{closure_env#0}>> #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\critical-section-1.1.2\src\lib.rs:226)
    8. <unknown function @ 0x10018284> #[inline] (c:\Users\Me\.cargo\registry\src\index.crates.io-6f17d22bba15001f\atomic-polyfill-1.0.3\src\polyfill.rs:134)
    9. allocate_alarm @0x0000000010018278 (c:\Users\Me\develop\MyFirmware\embassy\embassy-rp\src\timer.rs:47)
    10. _embassy_time_allocate_alarm @0x100191b4 (c:\Users\Me\develop\MyFirmware\embassy\embassy-time\src\driver.rs:162)
    11. allocate_alarm @0x1001997c (c:\Users\Me\develop\MyFirmware\embassy\embassy-time\src\driver.rs:134)
    12. <unknown function @ 0x100160e4> #[inline] (c:\Users\Me\develop\MyFirmware\embassy\embassy-executor\src\raw\mod.rs:335)
    13. <unknown function @ 0x100160e4> #[inline] (c:\Users\Me\develop\MyFirmware\embassy\embassy-executor\src\raw\mod.rs:496)
    14. <unknown function @ 0x100160e4> @0x00000000100160e0 (c:\Users\Me\develop\MyFirmware\embassy\embassy-executor\src\arch\cortex_m.rs:78)
    15. __cortex_m_rt_main @0x1000cac0 (c:\Users\Me\develop\MyFirmware\src\main.rs:35)
    16. __cortex_m_rt_main_trampoline @0x1000cab0 (c:\Users\Me\develop\MyFirmware\src\main.rs:35)
    17. <unknown function @ 0x100001e6> @0x100001e6 (Unknown Source:0)
    18. <unknown function @ 0x100001e6> @0x100001e6 (Unknown Source:0)
</details>

It seems to be trying to initialise the `alarm` in the `SyncExecutor`. It keeps trying to get the critical section for atomic processing and fails, resulting in an infinite loop. As far as I can tell,  [embassy/embassy-rp/src/critical_section_impl.rs line 49 to 68](https://github.com/embassy-rs/embassy/blob/95b3d9eb3b3657de3d7bc9c04f8fb83eae901640/embassy-rp/src/critical_section_impl.rs#L49) is looping.
I have nothing connected to the pins of the Pico except SWDIO, SWLCLK (and GND) for debugging.
The program is based on `examples/rp/src/bin/wifi_tcp_server.rs`, but I have modified it for my own use.
I have a feeling that this may be due to the settings in Cargo.toml, so I'm posting the dependencies.

```
[dependencies]
cortex-m = { version = "0.7.6", features = ["inline-asm"] }
cortex-m-rt = "0.7"
embedded-hal-1 = { package = "embedded-hal", version = "=1.0.0-rc.1" }
defmt = "0.3"
defmt-rtt = "0.4"
panic-probe = { version = "0.3", features = ["print-defmt"] }
embassy-executor = { version = "0.3.0", path = "embassy/embassy-executor", features = ["nightly", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-rp = { version = "0.1.0", path = "embassy/embassy-rp", features = ["defmt", "unstable-traits", "nightly", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-time = { version = "0.1.3", path = "embassy/embassy-time", features = ["nightly", "unstable-traits", "defmt", "defmt-timestamp-uptime"] }
embassy-net = { version = "0.1.0", path = "embassy/embassy-net", features = ["defmt", "nightly", "tcp", "udp", "dhcpv4", "medium-ethernet"] }
embassy-futures = { version = "0.1.0", path = "embassy/embassy-futures" }
cyw43 = { path = "embassy/cyw43", features = ["defmt", "firmware-logs"] }
cyw43-pio = { path = "embassy/cyw43-pio", features = ["defmt", "overclock"] }
static_cell = { version = "1.2.0", features = ["nightly"] }
embedded-io-async = { version = "0.5.0", features = ["defmt-03"] }
heapless = "0.7.16"
embedded-hal-async = { version = "1.0.0-rc.1", features = ["defmt-03"] }
```

I'm not even sure if this is an embassy issue, but if anyone else has had a similar situation, I'd be grateful for a solution.

Dirbaio, 2023-09-30:
This is probably #1736 / https://github.com/probe-rs/probe-rs/pull/1603

Dicklessgreat, 2023-09-30:
I unplugged Pico's USB to try it out, plugged it in while holding down BOOTSEL, and started debugging with VSCode, and it worked.
So, yes, this is the problem you mentioned.
I should have asked you sooner!
Thanks for your reply!

closed 2023-09-30 by Dicklessgreat