embassy-rs/embassy #1982: Suggest exposing `embassy_stm32::CRC::reconfigure` as public.

pmnxis, 2023-09-30:
While CRC is currently a widely used checksum mechanism, the current design makes it difficult to reuse it across different software modules with different `CrcConfig`. To address this, I propose making the `reconfigure` function public.

However, by exposing it, developers using embassy will need to handle mutual exclusion. While this may introduce some complexity, it's often unavoidable when sharing a single hardware module. Providing comments and examples in the documentation can help guide developers on proper usage.

https://github.com/embassy-rs/embassy/blob/4da97433173fcab5196e7c2a69c295edff3588eb/embassy-stm32/src/crc/v2v3.rs#L68:L93

xoviat, 2023-09-30:
> Okay: new rule for consistency: pub fn set_config and pub fn reconfigure are prohibited. impl SetConfig for X must be used.
there's some existing code that needs to be updated to reflect that.

okhsunrog, 2025-02-15:
@xoviat @jamesmunns I'm not sure I'm tagging the right people, but can we implement SetConfig for crc? I want to use it both for crc16 and crc32 in my project

kpfleming, 2025-05-26:
I'm willing to tackle this one if someone can provide a link to an example I can follow in another driver! Where did the 'new rule' quote above come from?