embassy-rs/embassy #1983: Can't build my program

Anivie, 2023-09-30:
Then I am try to build my program by `cargo build --release` , it will report these error:
```
error[E0425]: cannot find function `__basepri_r` in module `crate::asm::inline`
 --> D:\SDK\Rust\Cargo\regist
try\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri.rs:6:15
  |
6 |     call_asm!(__basepri_r() -> u8)
  |               ^^^^^^^^^^^ not found in `crate::asm::inline`
```
But I am already enable this feature:
`cortex-m = { version = "0.7.7", features = ["inline-asm", "critical-section-single-core"] }`

So how can I fix it?
This is my Cargo.toml
```
[package]
edition = "2021"
name = "embassy-stm32f1-examples"
version = "0.1.0"
license = "MIT OR Apache-2.0"

[dependencies]
embassy-stm32 = { git = "https://github.com/embassy-rs/embassy.git", features = ["nightly", "defmt", "stm32f103ve", "unstable-pac", "memory-x", "time-driver-any", "unstable-traits" ]  }
embassy-sync = { git = "https://github.com/embassy-rs/embassy.git", features = ["defmt"] }
embassy-executor = { git = "https://github.com/embassy-rs/embassy.git", features = ["nightly", "arch-cortex-m", "executor-thread", "defmt", "integrated-timers"] }
embassy-time = { git = "https://github.com/embassy-rs/embassy.git", features = ["defmt", "defmt-timestamp-uptime", "tick-hz-32_768"] }
embassy-usb = { git = "https://github.com/embassy-rs/embassy.git", features = ["defmt"] }
embassy-futures = { git = "https://github.com/embassy-rs/embassy.git" }

defmt = "0.3.5"
defmt-rtt = "0.4"

cortex-m = { version = "0.7.7", features = ["inline-asm", "critical-section-single-core"] }
cortex-m-rt = "0.7.3"
embedded-hal = "0.2.7"
panic-probe = { version = "0.3.1", features = ["print-defmt"] }
futures = { version = "0.3.28", default-features = false, features = ["async-await"] }
heapless = { version = "0.7.16", default-features = false }
nb = "1.1.0"

[profile.dev]
opt-level = "s"

[profile.release]
debug = 2
```

Dirbaio, 2023-09-30:
make sure you're using this exact nightly Rust version https://github.com/embassy-rs/embassy/blob/main/rust-toolchain.toml#L4

for example by copying that file into your project.

Anivie, 2023-09-30:
> make sure you're using this exact nightly Rust version https://github.com/embassy-rs/embassy/blob/main/rust-toolchain.toml#L4确保您使用的是这个精确的夜间 Rust 版本 https://github.com/embassy-rs/embassy/blob/main/rust-toolchain.toml#L4
> 
> for example by copying that file into your project.例如，将该文件复制到您的项目中。

I tried it but it seems that the problem is still not solved.
```
        Mode                 LastWriteTime          Length     Name
        ----                 -------------                   ------     ----
        d-----               2023/10/1      3:01                    .idea
        d-----               2023/10/1      3:01                    src
        d-----               2023/10/1      3:30                    target
        -a----               2023/10/1      2:41              8     .gitignore
        -a----               2023/10/1      1:57            171     build.rs
        -a----               2023/10/1      3:16          23722     Cargo.lock
        -a----               2023/10/1      3:30           1314     Cargo.toml
        -a----               2023/10/1      3:26            461     rust-toolchain.toml
```

Anivie, 2023-09-30:
I copied `Cargo.toml` and `build.rs` into a blank project and expected them to work, but my choice was wrong.

Since I don't know much about the rust build system, I didn't found that the default build target is switched in the example, which caused cargo tries to build the program for my running platform, when this error happend.

The build completed successfully after I provided the `--target thumbv7m-none-eabi` parameter manually.

closed 2023-09-30 by Anivie