embassy-rs/embassy #1993: Usb Serial example not working for stm32f3

Barbacamanitu, 2023-10-01:
I'm trying to get the serial usb example working for my STM32F3Discovery board. I'm running the example as is, except I'm omitting the lines

```
// Needed for nucleo-stm32f303ze
    let mut dp_pullup = Output::new(p.PG6, Level::Low, Speed::Medium);
    Timer::after(Duration::from_millis(10)).await;
    dp_pullup.set_high();
```

since I'm not using that chip. Once I flash the code to my board, I unplug my usb from the st-link port and plug it into the usb-user port. Windows gives me an error in device manager: 

```
Windows has stopped this device because it has reported problems. (Code 43)

A request for the USB device descriptor failed.
```

I've also tried doing some serial tests using this board with Stm32duino. I was able to get serial communication working with that. It seems like the chip is not providing a correct descriptor. Does anyone know why this might be?

Dirbaio, 2023-10-01:
can you post log output of the firmware with DEFMT_LOG=trace?

Barbacamanitu, 2023-10-01:
```
0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), apb1: Hertz(24000000), apb1_tim: Hertz(48000000), apb2: Hertz(24000000), apb2_tim: Hertz(48000000), ahb1: Hertz(48000000), adc: None, adc34: None }
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:130
0.000335 INFO Hello World!
└─ src\main.rs:47
0.200866 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.201324 TRACE   index=1 addr=64 len=8
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.201751 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.202178 TRACE   len_bits = 8400
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.202423 TRACE   index=2 addr=72 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.202850 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.203308 TRACE   index=2 addr=136 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.203735 INFO USB: device_descriptor used: 18
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.204010 INFO USB: config_descriptor used: 62
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.204254 INFO USB: bos_descriptor used: 12
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.204528 INFO USB: control_buf size: 7
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.204803 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.205200 TRACE   len_bits = 8400
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.205444 TRACE   index=0 addr=200 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.205841 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.206268 TRACE   index=0 addr=264 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.206665 TRACE enabled
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.207000 TRACE usb: power detected
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.207214 TRACE RESET
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.207611 TRACE usb: reset
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.207855 TRACE SETUP read waiting
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.208068 TRACE wait_enabled OUT WAITING
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.209777 TRACE usb: resume
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.210021 TRACE usb: suspend
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
```

Here you go. I only have one mini-usb cord at the moment though, so this won't include any information about me plugging into the usb-user port.

Dirbaio, 2023-10-01:
> so this won't include any information about me plugging into the usb-user port.

this is needed to debug the problem! :)

Barbacamanitu, 2023-10-02:
I'll order another usb cable and try that.

I tried disabling and re-enabling the USB device in device manager, and one time it worked. It showed up as COM7 and I was able to connect to it with Putty. But sending data had no effect. I haven't be able to get this to happen again. I keep getting errors about device descriptor or SetAddress failing.

I'll respond when I get a new usb cable. Thanks

Barbacamanitu, 2023-10-02:
I got it working... sort of. I adapted the code from here: https://github.com/embassy-rs/embassy/blob/main/examples/nrf52840/src/bin/usb_serial_winusb.rs

In the examples for stm32f3, there is only the usb_serial.rs. I didn't realize that windows has special requirements for recognizing usb devices.

Here is my full code currently: 
```
bind_interrupts!(struct Irqs {
    USB_LP_CAN_RX0 => usb::InterruptHandler<peripherals::USB>;
});

// This is a randomly generated GUID to allow clients on Windows to find our device
const DEVICE_INTERFACE_GUIDS: &[&str] = &["{EAA9A5DC-30BA-44BC-9232-606CDC875321}"];

#[embassy_executor::main]
async fn main(spawner: Spawner) {
    let mut config = Config::default();
    config.rcc.hse = Some(mhz(8));
    config.rcc.sysclk = Some(mhz(48));
    config.rcc.pclk1 = Some(mhz(24));
    config.rcc.pclk2 = Some(mhz(24));
    config.rcc.pll48 = true;
    let p = embassy_stm32::init(config);
    let mut leds = { Leds::new(p.PE9, p.PE8, p.PE10, p.PE15, p.PE11, p.PE14, p.PE12, p.PE13) };
    //let mut stepper1 = StepperMotor::new(p.PC3.degrade(), p.PC1.degrade(), 500, 2000, 200);

    // Create the driver, from the HAL.
    let driver = Driver::new(p.USB, Irqs, p.PA12, p.PA11);

    // Create embassy-usb Config
    let mut config = embassy_usb::Config::new(0xc0dd, 0xcaff);
    config.manufacturer = Some("Embassy");
    config.product = Some("USB-serial example");
    config.serial_number = Some("12345678");
    config.max_power = 100;
    config.max_packet_size_0 = 64;

    config.device_class = 0x02;

    // Create embassy-usb DeviceBuilder using the driver and config.
    // It needs some buffers for building the descriptors.
    let mut device_descriptor = [0; 256];
    let mut config_descriptor = [0; 256];
    let mut bos_descriptor = [0; 256];
    let mut msos_descriptor = [0; 256];
    let mut control_buf = [0; 64];

    let mut state = State::new();

    let mut builder = Builder::new(
        driver,
        config,
        &mut device_descriptor,
        &mut config_descriptor,
        &mut bos_descriptor,
        &mut msos_descriptor,
        &mut control_buf,
    );

    builder.msos_descriptor(windows_version::WIN10, 2);

    // Create classes on the builder.
    let mut class = CdcAcmClass::new(&mut builder, &mut state, 64);

    let msos_writer = builder.msos_writer();
    msos_writer.configuration(0);
    msos_writer.function(InterfaceNumber(0));
    msos_writer.function_feature(msos::CompatibleIdFeatureDescriptor::new("WINUSB", ""));
    msos_writer.function_feature(msos::RegistryPropertyFeatureDescriptor::new(
        "DeviceInterfaceGUIDs",
        msos::PropertyData::RegMultiSz(DEVICE_INTERFACE_GUIDS),
    ));

    // Build the builder.
    let mut usb = builder.build();

    // Run the USB device.
    let usb_fut = usb.run();

    // Do stuff with the class!
    let echo_fut = async {
        loop {
            class.wait_connection().await;
            info!("Connected");
            let _ = echo(&mut class).await;

            info!("Disconnected");
        }
    };

    // Run everything concurrently.
    // If we had made everything `'static` above instead, we could do this using separate tasks instead.
    join(usb_fut, echo_fut).await;
}
struct Disconnected {}

impl From<EndpointError> for Disconnected {
    fn from(val: EndpointError) -> Self {
        match val {
            EndpointError::BufferOverflow => panic!("Buffer overflow"),
            EndpointError::Disabled => Disconnected {},
        }
    }
}

async fn echo<'d, T: Instance + 'd>(
    class: &mut CdcAcmClass<'d, Driver<'d, T>>,
) -> Result<(), Disconnected> {
    let mut buf = [0; 64];
    loop {
        let n = class.read_packet(&mut buf).await?;
        let data = "test".as_bytes();
        //info!("data: {:x}", data);
        class.write_packet(data).await?;
    }
}
```

I say that I got it sort of working, because I have to take a few extra steps before I can communicate via serial. When I plug my dev board into my PC via the USB USER port, I get the same error as before:

```
Windows has stopped this device because it has reported problems. (Code 43)

A request for the USB device descriptor failed.
```

However, if I leave the board plugged in, then go to device manager and disable the USB device with the error, then enable it again, the error goes away. I then get a new USB Serial Device (COM7) appearing, but with an error 

```
This device cannot start. (Code 10)

A device which does not exist was specified.
```

If I disable and re-enable this device, I get a working USB Serial Device (COM7). I can open putty and connect to my board, and on every keystroke it sends back "test", so it seems to be working.

I'm at a loss for why this process seems to fix it. It seems to fix the issue every time. Obviously, it's far from idea. That's all the information I can give until I get the second usb cable.

Barbacamanitu, 2023-10-02:
I got a second usb cable at walmart. Here's the log:
```

0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), apb1: Hertz(24000000), apb1_tim: Hertz(48000000), apb2: Hertz(24000000), apb2_tim: Hertz(48000000), ahb1: Hertz(48000000), adc: None, adc34: None }
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:130
0.250732 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.251190 TRACE   index=1 addr=64 len=8
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.251647 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.252075 TRACE   len_bits = 8400
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.252319 TRACE   index=2 addr=72 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.252746 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.253173 TRACE   index=2 addr=136 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.256805 INFO USB: device_descriptor used: 18
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.257049 INFO USB: config_descriptor used: 62
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.257324 INFO USB: bos_descriptor used: 40
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.257568 INFO USB: msos_descriptor used: 178
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.257843 INFO USB: control_buf size: 64
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:143
0.258117 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.258544 TRACE   len_bits = 8400
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.258789 TRACE   index=0 addr=200 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.259155 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.259582 TRACE   index=0 addr=264 len=64
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.260009 TRACE enabled
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.260314 TRACE usb: power detected
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.260528 TRACE RESET
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.260925 TRACE usb: reset
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.261169 TRACE SETUP read waiting
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.261383 TRACE wait_enabled OUT WAITING
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
0.263122 TRACE usb: resume
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
0.263336 TRACE usb: suspend
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
6.249389 TRACE RESET
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.249786 TRACE usb: reset
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
6.250061 TRACE SETUP read waiting
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.282653 TRACE READ DONE, rx_len = 8
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.283020 TRACE SETUP read ok
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.283203 TRACE control request: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 256, index: 0, length: 64 }
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
6.283721 TRACE control: data_in
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.283905 TRACE WRITE WAITING
[... 287 lines cut ...]
6.425811 TRACE control: data_in
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.425994 TRACE WRITE WAITING
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.426544 TRACE WRITE OK
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.426818 TRACE SETUP read waiting
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.427032 TRACE READ DONE, rx_len = 8
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.427368 TRACE SETUP read ok
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.427581 TRACE control request: Request { direction: In, request_type: Standard, recipient: Device, request: 6, value: 770, index: 1033, length: 255 }        
└─ C:\dev\embedded\embassy\embassy-usb\src\fmt.rs:117
6.428405 TRACE control: data_in
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.428588 TRACE WRITE WAITING
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.429168 TRACE WRITE OK
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
6.429412 TRACE SETUP read waiting
└─ C:\dev\embedded\embassy\embassy-stm32\src\fmt.rs:117
```

Something strange happens when using both usb cables. I don't get any errors if I plug in the second usb cable once the board is running from the ST-Link usb port. If I plug in ONLY the USB USER port, I always get an error.  Every time I plug USER USB in after the other port is already active, I get no errors.

Resonanz, 2025-02-17:
Did you ever solve this problem?