embassy-rs/embassy #1998: pico w: adc example returns bogus temperature

kbleeke, 2023-10-02:
I don't know if I am missing something obvious but the temperature returns in the rp/adc example is very wrong

```
INFO  Pin 26 ADC: 4095
└─ adc::____embassy_main_task::{async_fn#0} @ src/bin/adc.rs:32
INFO  Pin 27 ADC: 4095
└─ adc::____embassy_main_task::{async_fn#0} @ src/bin/adc.rs:34
INFO  Pin 28 ADC: 4095
└─ adc::____embassy_main_task::{async_fn#0} @ src/bin/adc.rs:36
INFO  Temp: -44.0 degrees
└─ adc::____embassy_main_task::{async_fn#0} @ src/bin/adc.rs:38
```

It's not -44°C , the real value should be 25-ish according to my bme680

I am running this on a picow, is it not supposed to work on the picow? The adc to temperature conversion looks the same as

https://github.com/raspberrypi/pico-examples/blob/master/adc/onboard_temperature/onboard_temperature.c#L18

But I haven't run the C code yet.

Dirbaio, 2023-10-02:
the test is passing on a Pico W, perhaps check what you're doing differently in your firmware:

https://github.com/embassy-rs/embassy/blob/main/tests/rp/src/bin/adc.rs#L81-L82

kbleeke, 2023-10-02:
Looks like as soon as I connect my bme680 the measurement goes bad. Unfortunate but not an embassy problem

closed 2023-10-02 by kbleeke