embassy-rs/embassy #2024: Error code E0658 during cargo check

adoble, 2023-10-08:
I have done `cargo update` on my current project and since then get many `error[E0658]` messages. The project is for the Raspberry Pi Pico.

I can reproduce this error by setting up a new project using the code from the blinky example ( `examples/rp/src/bin/blinky.rd`) and the following `Cargo.toml` file:

```toml
[package]
name = "embassy-test"
version = "0.1.0"
edition = "2021"

[dependencies]
embassy-embedded-hal = { version = "0.1.0",  git = "https://github.com/embassy-rs/embassy", features = ["defmt",]   }

embassy-sync = { version = "0.3.0",  git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-executor = { version = "0.3.0",  git = "https://github.com/embassy-rs/embassy", features = ["nightly", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"] }
embassy-time = { version = "0.1.3",  git = "https://github.com/embassy-rs/embassy", features = ["nightly", "unstable-traits", "defmt", "defmt-timestamp-uptime"] }
embassy-rp = { version = "0.1.0",   git = "https://github.com/embassy-rs/embassy", features = ["defmt", "unstable-traits", "nightly", "unstable-pac", "time-driver", "critical-section-impl"] }
embassy-usb = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt"] }
embassy-futures = { version = "0.1.0",git = "https://github.com/embassy-rs/embassy" }

defmt = "=0.3.2"
defmt-rtt = "0.4"
fixed = "1.23.1"
fixed-macro = "1.2"

#cortex-m = { version = "0.7.6", features = ["critical-section-single-core"] }
#cortex-m = { version = "0.7.6", features = ["inline-asm"] }
cortex-m = "0.7.6"
cortex-m-rt = "0.7.0"
panic-probe = { version = "0.3", features = ["print-defmt"] }
futures = { version = "0.3.17", default-features = false, features = ["async-await", "cfg-target-has-atomic", "unstable"] }

embedded-hal-1 = { package = "embedded-hal", version = "=1.0.0-rc.1" }
embedded-hal-async = "1.0.0-rc.1"
embedded-hal-bus = { version = "0.1.0-rc.1", features = ["async"] }
embedded-io-async = { version = "0.6.0", features = ["defmt-03"] }
embedded-storage = { version = "0.3" }
static_cell = { version = "1.1", features = ["nightly"]}
log = "0.4"
pio-proc = "0.2"
pio = "0.2.1"
rand = { version = "0.8.5", default-features = false }

[profile.release]
debug = 2
```

And the error is:

```sh
Checking embassy-embedded-hal v0.1.0 (https://github.com/embassy-rs/embassy#9d311121)
error[E0658]: `async fn` return type cannot contain a projection or `Self` that references lifetimes from a parent scope
  --> C:\Users\T440s\.cargo\git\checkouts\embassy-9312dcb0ed774b29\9d31112\embassy-embedded-hal\src\flash\partition\asynch.rs:52:64
   |
52 |     async fn read(&mut self, offset: u32, bytes: &mut [u8]) -> Result<(), Self::Error> {
   |                                                                ^^^^^^^^^^^----^^^^^^^^
   |                                                                           |
   |                                                                           help: consider spelling out the type instead: `<Partition<'_, M, T> as embedded_storage_async::nor_flash::ReadNorFlash>`
   |
   = note: see issue #103532 <https://github.com/rust-lang/rust/issues/103532> for more information
   = help: add `#![feature(impl_trait_projections)]` to the crate attributes to enable

```
This is followed by more E0658 errors for other modules.

Note that:
```sh
$ rustc --version
rustc 1.72.0-nightly (8084f397c 2023-06-25)
```

I am aware that this may be due to some mistake in my own configuration, but I would appreciate it if anybody could provide a hint as to what I am doing wrong.

Dirbaio, 2023-10-08:
make sure you're using the embassy-supported nightly version: https://github.com/embassy-rs/embassy/blob/main/rust-toolchain.toml#L4

for example, copying that `rust-toolchain.toml` file into your project (deleting targets you don't need).

When you update Embassy, always check if the supported nightly version has changed, and update that too.

adoble, 2023-10-09:
Thanks @Dirbaio , that worked!

I didn't know about `rust-toolchain.toml` files, so I've also learnt something new. 

I'll close the issue.

closed 2023-10-09 by adoble