embassy-rs/embassy #2027: embassy-net: serving multiple connections?

esensible, 2023-10-09:
Firstly, I love your work. Thanks.

I've implement a simple, application specific httpd on the pico and want to use long-poll GET requests to synchronize multiple clients. As far as I can tell, embassy-net combines typical Berkley style accept/listen calls into a single accept call, which prevents the server moving clients to an ephemeral port when issuing the SYN-ACK. Am I misunderstanding the embassy-net API? As far as I can tell, smoltcp does not impose this restriction.

In the event that I'm misunderstanding, is there an embassy-net example demonstrating async handling of multiple, possibly overlapping network connections?

Thanks again.

Dirbaio, 2023-10-09:
yes, you can do it:

- create a task with `#[embassy_executor::task(pool_size=4)]`
- spawn the task 4 times. each copy does:
- loop
- create socket
- call accept() on port 80 -- it's fine to have N listening sockets on the same port, incoming connections go to a random one.
- handle the connection
- on disconnect, go back to loop.

(fwiw it works the same way in smoltcp. embassy-net accept does smoltcp listen + wait for a connection to actually arrive)

esensible, 2023-10-09:
Perfect. Thanks so much.
It being OK to have multiple listening sockets on the same port was unexpected, but awesome.

esensible, 2023-10-09:
It seems that on the rpi I need to choose between DHCP and another socket, even when increasing the resource count to Stack<>

esensible, 2023-10-09:
Thanks again. Your support was spot on. My being a noob rustacian corrupted by 30 years of C/C++ took a moment to get it going.

closed 2023-10-09 by esensible