embassy-rs/embassy #2029: STM32 Timer interrupt is invoked twice

ntqbit, 2023-10-09:
Hello.

I encountered a strange bug when testing timers on STM32F103C8T6 (Blue Pill). I'm not sure what caused the bug. However, since I could not reproduce it with stm32f1xx-hal crate (without embassy), I thought it would be reasonable not to exclude embassy as a potential cause.

The bug:
I configure the timer TIM3 (while apparently not being used by embassy) to generate an interrupt at a constant rate (once a second), a regular task.

But after I ran the program on the MCU I found that the interrupt is actually called two times (one right after the other) instead of one, at the configured rate (once a second).

Here is the interrupt code:
```
static mut OUTPUT: Option<Output<'static, embassy_stm32::peripherals::PA1>> = None;
static mut COUNTER: u32 = 0;

#[embassy_stm32::interrupt]
fn TIM3() {
    unsafe {
        COUNTER = (COUNTER + 1) % 4;
    }

    let output = unsafe { OUTPUT.as_mut().unwrap() };
    output.toggle();

    pac::TIM3.sr().modify(|w| w.set_uif(false));
}
```

The interrupt is configured to be invoked once a second.
Later in the `main`  there is a loop that outputs value of `COUNTER` once per 250ms (faster that the configured frequency of the timer):
    loop {
        defmt::info!("COUNTER: {}", unsafe { COUNTER });
        Timer::after(Duration::from_millis(250)).await;
    }

If the interrupt actually was called once per second, I would expect the output to be (incrementing by 1):
```
└─ src/main.rs:57   
0.500842 INFO COUNTER: 1
└─ src/main.rs:57       
0.751171 INFO COUNTER: 1
└─ src/main.rs:57       
1.001500 INFO COUNTER: 2
└─ src/main.rs:57       
1.251829 INFO COUNTER: 2
└─ src/main.rs:57       
1.502158 INFO COUNTER: 2
└─ src/main.rs:57       
1.752487 INFO COUNTER: 2
└─ src/main.rs:57       
2.002816 INFO COUNTER: 3
└─ src/main.rs:57       
2.253145 INFO COUNTER: 3
└─ src/main.rs:57       
2.503474 INFO COUNTER: 3
└─ src/main.rs:57       
2.753803 INFO COUNTER: 3
└─ src/main.rs:57       
3.004132 INFO COUNTER: 0
```

But instead, the output is incrementing by 2:
```
─ src/main.rs:56   
6.258209 INFO COUNTER: 2
└─ src/main.rs:56       
6.508538 INFO COUNTER: 2
└─ src/main.rs:56       
6.758867 INFO COUNTER: 2
└─ src/main.rs:56       
7.009196 INFO COUNTER: 0
└─ src/main.rs:56       
7.259508 INFO COUNTER: 0
└─ src/main.rs:56       
7.509820 INFO COUNTER: 0
└─ src/main.rs:56       
7.760132 INFO COUNTER: 0
└─ src/main.rs:56       
8.010444 INFO COUNTER: 2
└─ src/main.rs:56       
8.260773 INFO COUNTER: 2
└─ src/main.rs:56       
8.511102 INFO COUNTER: 2
└─ src/main.rs:56       
8.761431 INFO COUNTER: 2
└─ src/main.rs:56
```

I discovered the bug when I found that I could not set the output pin high by using `toggle` method, it always stayed low.
It makes sense why that happened if the interrupt is called twice.

The exact project that causes the bug is in [this repo](https://github.com/ntqbit/stm32-embassy-interrupt-problem).
You can just clone it and run with `DEFMT_LOG=info cargo embed debug`.

I confirmed that the bug happens also on my another MCU STM32F103C6T6A. The only difference between it and the blue pill is the Flash and RAM size, but, anyway.

If anybody have a STM32, does it happen to you as well?

What could be the problem? Why the timer interrupt might be called twice?

Dirbaio, 2023-10-09:
using `static mut` like that is UB. Can you try to reproduce it with `AtomicU32` / `critical_section::Mutex`? Just to rule it out.

ntqbit, 2023-10-10:
> using `static mut` like that is UB. Can you try to reproduce it with `AtomicU32` / `critical_section::Mutex`? Just to rule it out.

Well.. Now it works. However, many sorts of changes to the code "fix" the bug:
1) Disabling LTO optimization o codegen-units=1
2) Doubling the line "pac::TIM3.sr().modify(|w| w.set_uif(false));"
3) Placing the line "pac::TIM3.sr().modify(|w| w.set_uif(false));" before "output.toggle()"
4) Changing `COUNTER=(COUNTER+1)%4` to `COUNTER += 1`. In the disassembly view I see that the rust compiler adds an overflow check for later case.

The COUNTER is not the primary issue. I used it only to deduce that the interrupt was probably called multiple times and to illustrate the problem here.
The primary problem was with the `toggle` call on an output pin.

I've rewritten the code a little bit so it doesn't use static memory at all. Now toggling the pin is done with PAC.

```
#[embassy_stm32::interrupt]
fn TIM3() {
    // defmt::info!("TIM3");

    let odr = pac::GPIOA.odr().read().odr(1);
    if odr == pac::gpio::vals::Odr::LOW {
        // Set high
        pac::GPIOA.bsrr().write(|w| w.set_bs(1, true));
    } else {
        // Set low
        pac::GPIOA.bsrr().write(|w| w.set_br(1, true));
    }

    pac::TIM3.sr().modify(|w| w.set_uif(false));
}
```

Even now, the interrupt doesn't toggle the output pin. Or it does, but then immediately toggles it back. It is probably the case, but I can't confirm this, my cheap oscilloscope doesn't have enough resolution to capture toggle this fast.

However, uncommenting the defmt::info line fixes the problem and the pin begins to alternate as expected.
The changed code is [here](https://github.com/ntqbit/stm32-embassy-interrupt-problem/tree/v2) (branch v2).

timokroeger, 2023-10-10:
> 2. Doubling the line "pac::TIM3.sr().modify(|w| w.set_uif(false));"
> 3. Placing the line "pac::TIM3.sr().modify(|w| w.set_uif(false));" before "output.toggle()"

Wild guess: The timer runs from a slower clock and the interrupt flag takes multiple cycles to clear. The ISR returns before the flag is cleared and NVIC sees it as active, re-entering the ISR.

Did you have the exact same clock setup when you used the stm32f1xx-hal crate (without embassy)?

ntqbit, 2023-10-10:
> Did you have the exact same clock setup when you used the stm32f1xx-hal crate (without embassy)?

Indeed, I've rewritten the code to use stm32f1xx-hal literally line by line, and the problem persists. So, the issue is not with embassy :smile:

The code:
```
#![no_std]
#![no_main]

use cortex_m::peripheral::NVIC;
use cortex_m_rt::entry;
use defmt_rtt as _;
use panic_probe as _;
use stm32f1xx_hal::{
    pac::{self, interrupt},
    prelude::*,
};

#[interrupt]
fn TIM3() {
    // defmt::info!("TIM3 interrupt!");

    // Toggle PA1 output pin
    let gpioa = unsafe { &*pac::GPIOA::ptr() };
    let high = gpioa.odr.read().odr1().is_high();
    if high {
        // Set low
        gpioa.bsrr.write(|w| w.br1().set_bit());
    } else {
        // Set high
        gpioa.bsrr.write(|w| w.bs1().set_bit());
    }

    let tim3 = unsafe { &*pac::TIM3::ptr() };
    tim3.sr.write(|w| w.uif().clear_bit());
}

#[entry]
fn main() -> ! {
    defmt::info!("Start!");

    let cp = cortex_m::Peripherals::take().unwrap();
    let dp = pac::Peripherals::take().unwrap();

    // Configure clock
    let mut flash = dp.FLASH.constrain();
    let rcc = dp.RCC.constrain();

    let clocks = rcc.cfgr.freeze(&mut flash.acr);

    let rcc = unsafe { &*pac::RCC::ptr() };

    // Enable GPIOA
    rcc.apb2enr.modify(|_, w| w.iopaen().set_bit());

    // Configure PA1 as push-pull output.
    let gpioa = unsafe { &*pac::GPIOA::ptr() };
    gpioa.crl.modify(|_, w| {
        w.mode1().output();
        w.cnf1().push_pull();
        w
    });
    // Set PA1 low.
    gpioa.bsrr.write(|w| w.br1().set_bit());

    // Enable the TIM3 timer interrupt once a second.
    rcc.apb1enr.modify(|_, w| w.tim3en().set_bit());
    let tim3 = unsafe { &*pac::TIM3::ptr() };
    tim3.dier.write(|w| w.uie().set_bit());
    tim3.psc.write(|w| w.psc().variant(62500 - 1));
    tim3.arr.write(|w| w.arr().variant(128 - 1));
    tim3.cr1.write(|w| w.cen().set_bit());

    unsafe {
        NVIC::unmask(interrupt::TIM3);
    }

    loop {}
}
```

And again, uncommenting the defmt::info! line "fixes" the bug. 

Also I found that adding the following code TWICE (adding it once doesn't help) anywhere before `tim3.sr.write(|w| w.uif().clear_bit());` fixes the problem.
```
    if gpioa.crl.read().mode1().is_input() {
        unreachable!();
    }
```

Or, adding the same code ONCE after the `tim3.sr.write(|w| w.uif().clear_bit())` fixes the problem as well.

Resetting the UIF bit in the beginning of the interrupt seems to always help.
However, in the reference manual for STM32F1 I could not find any restrictions on when to reset the UIF bit.

I found multiple threads asking about the same problem, and it looks like your guess is correct.

> The timer runs from a slower clock and the interrupt flag takes multiple cycles to clear. The ISR returns before the flag is cleared and NVIC sees it as active, re-entering the ISR.

- https://community.st.com/t5/stm32-mcus-products/tim2-interrupt-fires-twice-when-optimize-speed-is-set/td-p/393273
- https://stackoverflow.com/questions/73841669/stm32-double-interrupt
- https://community.arm.com/support-forums/f/keil-forum/27967/external-irq-called-twice-depending-on-optimization-level
"Clear the interrupt flag as early as possible in your ISR. That way you increase your chance of the interrupt flag actually having been cleared before leaving the ISR. Otherwise you might catch the same interrupt again, which explains the "twice as high" values. To be on the safe side you might also want to consider inserting a synchronization barrier (DSB) before leaving the ISR to ensure that memory accesses are guaranteed to have completed"

closed 2023-10-10 by ntqbit

Dirbaio, 2023-10-10:
wow, what a fun issue! :sweat_smile: Glad you figured it out.