embassy-rs/embassy #2034: Not possible to use both ADCs with STM32H7

pbert519, 2023-10-10:
Hello,

i'm trying to use both ADC1 and ADC2 on a STM32H7A3 chip.
Unfortunately as soon as the second ADC is enabled (order does not matter) the other ADC is reseted.

I was able to trace it `new` function, which enables the RCC Clock and afterwards resets the peripheral.
```impl<'d, T: Instance> Adc<'d, T> {
    pub fn new(adc: impl Peripheral<P = T> + 'd, delay: &mut impl DelayUs<u16>) -> Self {
        embassy_hal_internal::into_ref!(adc);
        T::enable();
        T::reset();
```
In this case, both ADCs share the reset:
`RCC_AHB1RSTR::ADC12RST`
and the already initalised ADC is reseted when the new() function is called for the second ADC.
The RccPeripheral trait is implemented as part of the generated stuff.

I'm open to fix this issue myself, however i'm not familiar with the metapac process.
Are they any similar Cases for other peripherals? How are you usually handle it, if the generated code does not fit?
Any hints in the right direction are appreciated.

Dirbaio, 2023-10-10:
ADC sharing RCC bits is very annoying. It's not only H7, other families have that too :(

embasy-stm32 tries to handle it by refcounting uses of peripherals that share RCC bits. The code is [here](https://github.com/embassy-rs/embassy/blob/eff73d6dfa4c5920a55a5ee2bf5c0b2ef68fbae1/embassy-stm32/build.rs#L407) . However this doesn't account for reset :smiling_face_with_tear: 

I think the fix is changing the RccPeripheral API to have `reset_and_enable(), disable()`. No separate `reset()`. Then `reset_and_enable` would only do the reset when refcount goes from 0 to 1.

pbert519, 2023-10-10:
Thanks for the response.

I greped the code base for usages of `T::reset()`, and already found a usecase [(usart](https://github.com/embassy-rs/embassy/blob/eff73d6dfa4c5920a55a5ee2bf5c0b2ef68fbae1/embassy-stm32/src/usart/buffered.rs#L175), ) where it's not simply possible to replace `T::enable()` and `T::reset()` with `T::enable_and_reset()`
However we could call `enable_and_reset()` twice here, the other usages in embassy-stm32 can be simply migrated.
However the other usages in embassy-stm32 could be migrated.

The [adc v3](https://github.com/embassy-rs/embassy/blob/eff73d6dfa4c5920a55a5ee2bf5c0b2ef68fbae1/embassy-stm32/src/adc/v3.rs#L15C3-L15C3) does not use the `reset()` and `enable()` at all, but implemeted the logic on it's own.
Seems to be the workaround for the same issue.

I'll try to the implement the `enable_and_reset()` functionality

Dirbaio, 2023-10-10:
> The [adc v3](https://github.com/embassy-rs/embassy/blob/eff73d6dfa4c5920a55a5ee2bf5c0b2ef68fbae1/embassy-stm32/src/adc/v3.rs#L15C3-L15C3) does not use the reset() and enable() at all, but implemeted the logic on it's own.
> Seems to be the workaround for the same issue.

yes, that code is from before we had refcounting for ADC RCC. It could be changed to RccPeripheral now.

PR #2035 (2023-10-12): STM32: combine RccPeripherals reset() and enable() to enable_and_reset()
Close  #2034

Combining the standalone functions `reset()` and `enable()` into `enable_and_reset()`.
 
 This are a lot of breaking changes, maybe we should add `reset_and_enable()`, but keep `enable()` and `reset()`. 
 And port only the obvious usages.
 I'm not able to test everything on real hardware.
closed 2023-10-12 by Dirbaio