embassy-rs/embassy #2048: STM32: TimeoutI2c doesn't appear to implement the async I2c traits

Ralim, 2023-10-12:
Hello,
I'm looking to use the `TimeoutI2c` wrapper  with a function that is setup to want the `I2c` traits (for async I2C).
However I've been running into
`the trait `embedded_hal_async::i2c::I2c` is not implemented for `TimeoutI2c<'_, '_, _, _, _>`

I _believe_ looking at `timeout.rs` that this is just a case of this not explicitly implementing this trait.

Would be amazing if someone could confirm (A) I'm not missing something obvious and (B) if just explicitly implementing this trait would fix this.

I'm happy to open a PR to reshuffle this file if (B) is correct 😁

lulf, 2023-10-12:
Reopen if you see issues (and kudos to @andresovela for fixing)

closed 2023-10-12 by lulf

Ralim, 2023-10-12:
Amazing; thank you 🙇🏼 
Super appreciate how fast this was fixed ❤️