embassy-rs/embassy #2052: Cannot compile for chip stm32g070cb Error: p.TIM3 error[E0609]: no field `TIM3` on type `Peripherals`

smeenka, 2023-10-13:
I try to use p.TIM1 and p.TIM3 for pwm generation.

My board has chip STM32G070CBTx which is for embassy feature "stm32g070cb"

Setting the feature features =  "stm32g070cb"   on [dependencies.embassy-stm32] gives compile error  p.TIM3 error[E0609]: no field `TIM3` on type `Peripherals`

Setting the feature to "STM32g071cb" does let me compile the code. The code is operational: I can control the motors connected to TIM3, The motors connected to TIM1 do not react. Of course I do understand the chip is incorrect now

Diving into the pac definitions of both chips in stm32-data-generated I see that for both chips TIM1 and TIM3 are defined, even with the same chip register addresses.

So my request: What is the problem here?

Dirbaio, 2023-10-13:
Probably TIM3 is being used for the embassy-time driver. Remove feature `time-driver-any`, add another `time-driver-timX` to make it use another timer.

ExplodingWaffle, 2023-10-13:
(and the issue with TIM1 is probably #2053)

smeenka, 2023-10-14:
Selecting feature time-driver-tim15 did let me compile for stm32g070cb.
After updating the main branch  of embassy the motors on TIM1 and TIM3 are functional now. Thanks!

Request: can there be a few comment lines in the cargo.toml of embassy-stm32 about what features are possible (and resolved dynamically?
Or even better: a link in the toml file to a markdown file with all possible features (from the user point of view, statically and dynamical) with a short description?

Embassy is really great, I really like it. Switched from rtic to embassy. But the low level stuff is hard, very hard. For example: If you want to give a module the ownership of adc, what is the typing of the adc parameter in this case. Up to today I was not able to figure this out. No example to help me out.

Documentation can make life a bit more easy!
You can close this issue.

closed 2025-01-10 by Dirbaio