embassy-rs/embassy #2065: Something fishy with crates.io embassy-time version

dossalab, 2023-10-15:
Hey,

I know that I (and many other people as well) asked about `crates.io` repository before, but this issue is a bit different because the crate `embassy-time` seems to be updated there. Attempting to use it I mentioned a weird artifact - all of my timings are stretched x30, i.e instead of 50 milliseconds I'm getting around ~1.5 seconds.

Importing it like that yields this issue:
```
embassy-time = { version = "0.1.4", features = [ "nightly" ] }
```

Using *the same version* pulled from GitHub repo works:
```
[dependencies.embassy-time]
version = "0.1.4"
git = "https://github.com/embassy-rs/embassy"
features = [ "nightly" ]
```

I'm not sure why it's like that but I spent too many time figuring it :(
It's hard to tell which exact tag cargo is using in both cases....

xoviat, 2023-10-15:
The crates.io version is almost completely unsupported. If you find a problem with it, there is almost no way to fix it other than to upload a new version.

dossalab, 2023-10-15:
> The crates.io version is almost completely unsupported. If you find a problem with it, there is almost no way to fix it other than to upload a new version.

But why does the same semver produce different results? UI shows that it was uploaded 3 days ago

Qix-, 2023-10-16:
@dossalab What xoviat is true, use the `git = "..."` directive in Cargo.toml for pulling in Embassy. Don't use crates.io.

For example,

```toml
[dependencies.embassy-executor]
git = "https://github.com/embassy-rs/embassy.git"
features = [
	"nightly",
	"integrated-timers",
	"executor-thread"
]

[dependencies.embassy-time]
git = "https://github.com/embassy-rs/embassy.git"
features = [
	"nightly",
	"unstable-traits",
	"tick-hz-32_768"
]
```

Dirbaio, 2023-10-16:
the root cause of this issue is you have multiple copies of embassy-time in your dependency graph, likely one from git and another from crates.io. Cargo enforces there's only one copy as long as you stick to the same source (like, all crates.io). If you're mixing, you're not protected. You can check this is the case with `cargo tree`.

Each copy of the crate has its own "tick rate", so you can get situations like the HAL depends on embassy-time from git and is setting one tick rate, while you're depending on embassy-time from crates.io and setting another tick rate. So when you sleep, seconds are converted to ticks using one rate, and the driver sleeps using the other, resulting in the mismatch.

embassy-time v0.1.5 has been just released, which adds a `links` key to `Cargo.toml` to enforce there's no multiple copies of the crate even across sources. Run `cargo update` and update your git commit hash to latest. If you get errors relating to multiple copies you'll have to fix them. When you do you'll be guaranteed no more tick rate mismatches.

closed 2023-10-16 by Dirbaio