embassy-rs/embassy #2072: STM32: Possible to have async uart with only one DMA

Ralim, 2023-10-16:
Hello again,
I'm using the `stm23g070cb` which only has `DMA1` with 8 DMA channels.

I would like to be able to do `async` receive via DMA on three UART's. While also using DMA channels for other devices.
I dont need `async` transmit in this instance.

Is it possible for the `impl` for `embedded_io_async::Write` and `embedded_io_async::Read` to depend on their respective DMA channel? I believe my current read of this is that both are required for the `impl` to apply.

This would allow for transmit only or receive only dma accelerated async implementations I believe?

Dirbaio, 2023-10-16:
This is already supported, you can directly create an UartRx or UartTx directly, which will only consume 1 DMA channel.

Ralim, 2023-10-16:
What is the incantation for mixed DMA Rx and non DMA Tx for example?
I can't for example do

```
let uart_demo_tx = UartTx::new(
        p.USART2,
        p.PA2,
        NoDma,
        embassy_stm32::usart::Config::default(),
    );

    let uart_demo_rx = UartRx::new(
        p.USART2,
        IrqsUSART2,
        p.PA3,
        NoDma,
        embassy_stm32::usart::Config::default(),
    );
```

As of course p.USART2 has already been moved by the first `UartTx`.

Dirbaio, 2023-10-16:
> What is the incantation for mixed DMA Rx and non DMA Tx for example?

ahh, *that* is not possible currently, unfortunately.

I think it should be easy to add though. The API could look like: create a non-DMA uart, split it, then "upgrade" one of the halves to have DMA.

Ralim, 2023-10-17:
Ah darn it; can't let it be to easy then 😓 
I was quite hopefuly it was me messing up.

I would be definitely happy with that API; It might be more intuitive to implement the interfaces for the cases of only having one DMA and one NoDma if possible (rather than split + upgrade). But I would take any API to let me get away with the limited 8 DMA channels 😁 

Thank you for taking the time to look at this btw ❤️

Ralim, 2024-06-04:
Hia,

Up until now I've been able to make this work using the pattern of making the uart using `NoDma` for the tx channel and then splitting the UART and upgrading the rx to a ring buffered UART.

i.e.
```rust
let uart = Uart::new(peripheral, rx_pin, tx_pin, irq_item, NoDma, rx_dma_item, uart_config).unwrap();
let (tx, rx) = uart.split();
let rx_buffered = rx.into_ring_buffered(ring_buffer);
// Then use tx and rx_buffered
```

I've run into this no longer being possible after the refactors recently to remove the peripheral generic items.
(as `NoDma` can no longer be used). I also cant see any method to 'upgrade' to add a dma being possible so far 😓

techmccat, 2025-05-22:
An upgrade method like 
```rust
fn upgrade(self, rx_dma: Peri<'d, impl RxDma<T>>) -> UartRx<'d, Async>
```
would require knowing the type of the underlying peripheral to put bounds on the DMA channel, which isn't possible anymore since commit 183f2f6913032600d74ea058b50a1fcedbebe719 moved the instance type parameters from the struct to the constructor.

@Dirbaio since this doesn't appear to be a trivial fix, you might want to reopen this issue.

If the type erasure is mandatory my proposed solution is to add a builder that keeps the instance type parameter, like so

```rust
struct<'d, T: Instance> UartRxBuilder { ... }
impl<'d, T: Instance> UartRxBuilder<'d, T> {
    fn build_blocking(self) -> UartRx<'d, Blocking> { ... }
    fn build_async(self, rx_dma: Peri<'d, impl RxDma<T>>) -> UartRx<'d, Async> { ... }
}
```

PR #2739 (2024-03-27): Provide embedded_io impls for Uart with and without Dma
Weakens the requirement `TxDma: self::TxDma<T>` in the implementation of `embedded_io::Write` for both `Uart` and `UartTx`.

Provides implementations for `embedded_io::Read` for `Uart` and `UartRx`, as well as `embedded_io_async::Read`, bringing it to parity with `BufferedUart`.

These changes seem to be sufficient to make the blocking and async APIs available together in mixed scenarios, like in #2072.

Fixes #2072