embassy-rs/embassy #2079: Wrong memory.x in example for nrf52840

michaelrommel, 2023-10-17:
Dear all,

while trying to compile the blinky example for a Seeed XIAO BLE, I stumbled over the provided `memory.x` file, which has a `RAM:` line which does not cater for the softdevice, although the softdevice is explicitly mentioned. 

See: https://github.com/embassy-rs/embassy/issues/1275#issuecomment-1465283792 for a reference how to fix this.

The cuplrit is at least in the `RAM:` statement, where the 8 byte offset and smaller size is needed: https://github.com/embassy-rs/embassy/blob/683d5c30669bbb788e60ee3dd31ce30ba14c2d69/examples/nrf52840/memory.x#L6

But I think that the flash origin should be adapted to `0x27000` if the softdevice is in the flash.

I have not enough experience to make a definitive suggestion here, I can only reference, what worked for me for the XIAO BLE.

I would appreciate if either a comment statement is added to the provided `memory.x` file or some bits and pieces are written about that in the documentation. This caused several hours of wasted time trying to debug a simple blinky example, which should work out of the box for inexperienced users...

Many thanks and if I get more experience I hope in the future I can submit PRs... Michael.

lulf, 2023-10-17:
I understand it can be a bit confusing! The examples are not meant to be run with softdevice, but the comment in the memory.x is anyway a bit misleading. How about we modify it to the following, with the assumption that softdevice RAM wil generally be below 128K (a warning is printed anyway at runtime if you have spare RAM so you know how to adjust it).

```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  FLASH : ORIGIN = 0x00000000, LENGTH = 1024K
  RAM : ORIGIN = 0x20000000, LENGTH = 256K

  /* These values correspond to the NRF52840 with Softdevices S140 7.3.0 */
  /*
     FLASH : ORIGIN = 0x00027000, LENGTH = 868K
     RAM : ORIGIN = 0x20020000, LENGTH = 128K
  */
}

```

michaelrommel, 2023-10-17:
I think this would be sufficient - let me try these values on my device later and report back, whether they work. I used the `ORIGIN = 0x20000008, LENGTH = 0x3fff8` line from Dirbaio's comment. I'll report back here later!

Thanks for looking into it!

lulf, 2023-10-17:
What ORIGIN you choose for RAM depends on if you enable the softdevice or not in your app. I guess for these examples you don't so it's ok with those settings, but perhaps defaulting to a higher offset will cause less problems later.

PR #2091 (2023-10-19): docs: add linker script comments
Existing comment were outdated. Provide an example configuration for using the softdevice with the nRF52 examples.

Fixes #2079
closed 2023-10-19 by lulf