embassy-rs/embassy #2100: Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor.

PhaestusFox, 2023-10-22:
I don't know what I did but my Pico not just crashes with this error any time I try to use a Timer or Ticker, this was working fine this afternoon but stopped after a cargo update.

I have made a minimal example to recreate it:
cargo.toml
```
[package]
name = "time_test"
version = "0.1.0"
edition = "2021"

# See more keys and their definitions at https://doc.rust-lang.org/cargo/reference/manifest.html

[dependencies]
embassy-executor = { version = "0.3.0", git = "https://github.com/embassy-rs/embassy", features = ["nightly", "arch-cortex-m", "executor-thread", "executor-interrupt", "defmt", "integrated-timers"]  }
embassy-time = { version = "0.1.5", git = "https://github.com/embassy-rs/embassy", features = ["nightly", "unstable-traits", "defmt", "defmt-timestamp-uptime"]  }
embassy-rp = { version = "0.1.0", git = "https://github.com/embassy-rs/embassy", features = ["defmt", "unstable-traits", "nightly", "unstable-pac", "time-driver", "critical-section-impl"]  }
defmt-rtt = "0.4"
panic-probe = { version = "0.3", features = ["print-defmt"] }
cortex-m-rt = "0.7.0"
cortex-m = { version = "0.7.6", features = ["inline-asm"] }
```

main.rs
```
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]
use embassy_executor::Spawner;
use embassy_time::{Timer, Duration};
use {defmt_rtt as _, panic_probe as _};
use embassy_rp::{bind_interrupts, pio::InterruptHandler, peripherals::PIO0};

bind_interrupts!(struct Irqs {
    PIO0_IRQ_0 => InterruptHandler<PIO0>;
});

#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    Timer::after(Duration::from_millis(10)).await
}
```

also have memory.x and build.rs from the pi examples

here is the full error message when I run this using a Pico probe:
```
0.000538 ERROR panicked at 'Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor.'
└─ embassy_executor::raw::waker::task_from_waker @ C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\fmt.rs:106 
0.000596 ERROR panicked at C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-0.3.5\src\lib.rs:368:5:
explicit panic
└─ panic_probe::print_defmt::print @ C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\panic-probe-0.3.1\src\lib.rs:104
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
Frame 0: HardFault @ 0x1000283c
Frame 1: __udf @ 0x10000ca2 inline
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\..\asm/inline.rs:181:5
Frame 2: udf @ 0x0000000010000ca2 inline
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src/call_asm.rs:11:43
Frame 3: hard_fault @ 0x0000000010000ca2 inline
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\panic-probe-0.3.1\src/lib.rs:86:5
Frame 4: panic @ 0x0000000010000ca2
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\panic-probe-0.3.1\src/lib.rs:54:9
Frame 5: panic_fmt @ 0x10001832
       /rustc/1c05d50c8403c56d9a8b6fb871f15aaa26fb5d07\library/core/src/panicking.rs:72:14
Frame 6: panic_display<&str> @ 0x10001890 inline
       /rustc/1c05d50c8403c56d9a8b6fb871f15aaa26fb5d07\library/core/src/panicking.rs:178:5
Frame 7: panic_explicit @ 0x0000000010001872
       /rustc/1c05d50c8403c56d9a8b6fb871f15aaa26fb5d07\library/core/src/panicking.rs:160:5
Frame 8: panic_cold_explicit @ 0x10001632
       C:/rustc/1c05d50c8403c56d9a8b6fb871f15aaa26fb5d07/library/core/src/panic.rs:87:13
Frame 9: default_panic @ 0x10001624
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-0.3.5\src/lib.rs:368:5
Frame 10: panic @ 0x100013ca inline
       C:\Users\Phox\.cargo\registry\src\index.crates.io-6f17d22bba15001f\defmt-0.3.5\src\export/mod.rs:133:14
Frame 11: task_from_waker @ 0x00000000100013c6 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src/fmt.rs:106:13
Frame 12: schedule_wake @ 0x00000000100013b4 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/mod.rs:600:20
Frame 13: _embassy_time_schedule_wake @ 0x00000000100013b4
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-time\src/queue.rs:55:13
Frame 14: schedule_wake @ 0x10001492 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-time\src/timer.rs
Frame 15: poll @ 0x000000001000148c
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-time\src/timer.rs:113:13
Frame 16: {async_fn#0} @ 0x10000244 inline
       G:\RustStuff\Projects\dry_box\time_test\src/main.rs
Frame 17: <unknown function @ 0x10000244> @ 0x0000000010000240
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/mod.rs:161:15
Frame 18: <unknown function @ 0x10001252> @ 0x10001252 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/util.rs:55:9
Frame 19: <unknown function @ 0x10001252> @ 0x0000000010001252 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/timer_queue.rs:34:28
Frame 20: {closure#0} @ 0x0000000010001252 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/mod.rs:418:17
Frame 21: <unknown function @ 0x10001252> @ 0x0000000010001238 inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/run_queue.rs:85:13
Frame 22: <unknown function @ 0x10001252> @ 0x000000001000121a inline
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/mod.rs:391:13
Frame 23: <unknown function @ 0x10001252> @ 0x0000000010001178
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\raw/mod.rs:532:9
Frame 24: <unknown function @ 0x100002dc> @ 0x100002dc
       C:\Users\Phox\.cargo\git\checkouts\embassy-9312dcb0ed774b29\e70c531\embassy-executor\src\arch/cortex_m.rs:107:21
Frame 25: __cortex_m_rt_main @ 0x1000041a
       G:\RustStuff\Projects\dry_box\time_test\src/main.rs:13:1
Frame 26: __cortex_m_rt_main_trampoline @ 0x10000404
       G:\RustStuff\Projects\dry_box\time_test\src/main.rs:13:1
Frame 27: <unknown function @ 0x100001e6> @ 0x100001e6
Frame 28: <unknown function @ 0x100001e6> @ 0x100001e6
Error: CPU halted unexpectedly.
error: process didn't exit successfully: `probe-rs run --chip RP2040 G:\RustStuff\Projects\dry_box\target\thumbv6m-none-eabi\release\time_test` (exit code: 1)
```

Verequies, 2023-10-22:
I have also encountered this issue. It is fixed if you downgrade to a prior version of Rust nightly. Seems like something changed in the last few days has caused Embassy to crash on start on the RP2040.

DCNick3, 2023-10-25:
This was reported to rustc as https://github.com/rust-lang/rust/issues/117047

AzazKamaz, 2023-10-25:
Several alternative ways to fix, in file `embassy-executor/src/raw/waker.rs`:
- make `VTABLE` `static` instead of `const` - probably proper way to fix
- make `VTABLE` functions `inline(never)` - `clone()`, `wake()`, `drop()`
- make functions that refer `VTABLE` `inline(never)` - `clone()`, `from_task()`, `task_from_waker()`

Also you can just disable cross-crate inlining with `-Zcross-crate-inline-threshold=0` compiler option

Dirbaio, 2023-10-25:
fixed in #2112

commit 0cc3e18d (2023-10-25): Merge pull request #2112 from AzazKamaz/patch-1

Fix #2100 - function address comparison
commit 6b19c0ab (2023-10-25): Fix #2100 - function address comparison
closed 2023-10-25 by Dirbaio