embassy-rs/embassy #2108: Embassy code won't compile now. Compiled a couple of weeks ago

Barbacamanitu, 2023-10-23:
I haven't touched my code in a couple of weeks. I was about to make some changes to incorporate a new crate I wrote, and I thought I'd start by just running my old code. However, my unchanged code won't compile. I'm getting errors about inline asm not being able to find some functions.

The errors: 
`error[E0425]: cannot find function `__basepri_r` in module `crate::asm::inline`                                                                                       
 --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri.rs:6:15
  |
6 |     call_asm!(__basepri_r() -> u8)
  |               ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_w` in module `crate::asm::inline`
  --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri.rs:22:19
   |
22 |         call_asm!(__basepri_w(basepri: u8));
   |                   ^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__basepri_max` in module `crate::asm::inline`
  --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\basepri_max.rs:19:19
   |
19 |         call_asm!(__basepri_max(basepri: u8));
   |                   ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error[E0425]: cannot find function `__faultmask_r` in module `crate::asm::inline`
  --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\register\faultmask.rs:29:28
   |
29 |     let r: u32 = call_asm!(__faultmask_r() -> u32);
   |                            ^^^^^^^^^^^^^ not found in `crate::asm::inline`

error: invalid register `r0`: unknown register
   --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\..\asm\inline.rs:197:24
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));
    |                        ^^^^^^^^^^^^^^

error: invalid register `r1`: unknown register
   --> C:\Users\russm\.cargo\registry\src\index.crates.io-6f17d22bba15001f\cortex-m-0.7.7\src\..\asm\inline.rs:197:40
    |
197 |     asm!("bkpt #0xab", inout("r0") nr, in("r1") arg, options(nomem, nostack, preserves_flags));`

I was using a cloned embassy repository and specifying my embassy dependency using the local path. What could have changed to cause this issue?

sureshjoshi, 2023-10-23:
Maybe a silly question - but this is what I often get when I'm not compiling with the correct target. Are you using the correct one?

Barbacamanitu, 2023-10-24:
I have my rusttoolchain.toml still from the last time it compiled. I have my .cargo folder with the config.toml set up correctly. Same as before. Is there something else I might be missing?

Dirbaio, 2023-10-24:
the error is definitely incorrect target. you need to either pass `--target` to Cargo, or set it in `.cargo/config.toml`.

Barbacamanitu, 2023-10-24:
I had to change my workspace level resolver to resolver = "2". If I moved my embassy related code out of my workspace, it compiled. I must have combined projects into a workspace at some point and forgotten.

They really should fix the fact that individual projects can have edition or resolver set to a specific value, but its ignored when those projects are in a workspace. I'm developing a local crate alongside my embedded firmware, and I need to be able to move between them and debug, so they're in the same workspace.

closed 2023-11-04 by lulf