embassy-rs/embassy #2114: Multi-Target Codebase Example

the-wondersmith, 2023-10-25:
I'm working on a project that absolutely should be able to be written (more or less) in a compilation target agnostic fashion. I *think* I've got maybe ~65% of it worked out, but I'm kinda stuck and think I'm too deep in the trees to be able to see the forest clearly.

I've pushed up an [example repo](https://github.com/the-wondersmith/embassy-multi-target-example) as a sort of "visual aid". The end goal is to simply blink the on-board LED when the target _has_ an one, or to "emulate" the LED blink with a simple log like `LED turned on` / `LED turned off`.

The project is not currently functional, as I cannot figure out how to handle hardware abstraction and (potentially) the `std` vs `no_std` compilation. Could I possibly beg some help here?

tomasbarton-stemcell, 2023-10-31:
I did this by splitting the project into multiple crates where the hardware independent logic is in a separate no_std crate (or more crates if this makes sense for your project). This crate defines traits for various HW pieces it needs (in your example it would be `Led` with two methods `on()` and `off()`). It also defines a "main" function that takes generics bounded with these traits.

Then you would create a separate crate for each target lets say stm32, std, etc. These crates would use the hw independent crate as a dependency and implement the traits you defined earlier (stm by toggling a pin, std by printing into a log), instantiate the structs and pass them to the main function mentioned above.

I found that having a crate per architecture is much nicer than using conditional compilation. You will have a few more files, but on the other hands those files will be much cleaner.

I also use a Cargo workspaces: I have all hardware independent crates in a single workspace. This runs unit tests, etc. And then I have the target dependent crates outside of this workspace. I found that Cargo workspace does not work that great when you have multiple different targets. I do not run unit tests in the hardware dependent crates (that's why I try to factor out as much code out of these crates and have them only to initialize the hardware and pass the control). The reason for this is that when Cargo compiles your code it will attempt to unify all features which often results in conflicts if you have target specific features enabled together with your test dependencies (that would use std).

This article describes a similar architecture: https://ferrous-systems.com/blog/test-embedded-app/ It focuses on testing, but the proposed structure is very similar to what I described.

PhilippMolitor, 2024-10-11:
You might find this interesting:
https://github.com/PhilippMolitor/embassy-multitarget

I implemented crates for each target MCU, which just bootstrap the entrypoint. Tasks are defined in seperate crates, and the peripherals are passed with function arguments that only contain trait definitions from `embedded-hal` and `embedded-hal-async`. Please note some stuff is very much work in progress, but it is a good starting point. 

Also, `embassy-executor` must be a workspace dependency, as its macros for `main` and `task` expand to something using `::embassy_executor`, which only works when the crate is a direct dependency of the crate using it.

Token-Thinker, 2024-10-12:
@PhilippMolitor 

We got pretty far, however some issues with the different HALs when trying to build the various crates.

Take a look here -> https://github.com/Token-Thinker/rusty-robot

closed 2025-01-10 by Dirbaio