embassy-rs/embassy #2153: example stm32f1 usb_serial: unable to enumerate USB device

nim65s, 2023-11-06:
Hi,

On the current `main` branch, when I run `cargo run --bin usb_serial --release` in the `embassy/examples/stm32f1` folder, it correctly build and flash my bluepill (stm32f103c8) through a stlink v2:

```
$ cargo run --bin usb_serial --release
    Finished release [optimized + debuginfo] target(s) in 0.25s
     Running `probe-rs run --chip STM32F103C8 target/thumbv7m-none-eabi/release/usb_serial`
     Erasing sectors ✔ [00:00:02] [########################################################] 37.00 KiB/37.00 KiB @ 13.73 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:55] [#########################################################] 37.00 KiB/37.00 KiB @    680 B/s (eta 0s )    Finished in 58.348s
0.000000 TRACE BDCR configured: 00000200
└─ embassy_stm32::rcc::bd::{impl#2}::init @ …/embassy/embassy-stm32/src/fmt.rs:117 
0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), pclk1: Hertz(24000000), pclk1_tim: Hertz(48000000), pclk2: Hertz(48000000), pclk2_tim: Hertz(48000000), hclk1: Hertz(48000000), adc: Some(Hertz(6000000)), rtc: Some(Hertz(40000)) }
└─ embassy_stm32::rcc::set_freqs @ …/embassy/embassy-stm32/src/fmt.rs:130 
0.000000 INFO  Hello World!
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src/bin/usb_serial.rs:30  
```

But the device is not recognized on linux. `dmesg` show the following:
```
[82148.285782] usb 1-6.4.4.2: new full-speed USB device number 43 using xhci_hcd
[82148.382465] usb 1-6.4.4.2: device descriptor read/64, error -32
[82148.585779] usb 1-6.4.4.2: device descriptor read/64, error -32
[82148.789064] usb 1-6.4.4.2: new full-speed USB device number 44 using xhci_hcd
[82148.885799] usb 1-6.4.4.2: device descriptor read/64, error -32
[82149.089093] usb 1-6.4.4.2: device descriptor read/64, error -32
[82149.196025] usb 1-6.4.4-port2: attempt power cycle
[82149.835745] usb 1-6.4.4.2: new full-speed USB device number 45 using xhci_hcd
[82149.836137] usb 1-6.4.4.2: Device not responding to setup address.
[82150.042773] usb 1-6.4.4.2: Device not responding to setup address.
[82150.249053] usb 1-6.4.4.2: device not accepting address 45, error -71
[82150.345714] usb 1-6.4.4.2: new full-speed USB device number 46 using xhci_hcd
[82150.346162] usb 1-6.4.4.2: Device not responding to setup address.
[82150.552830] usb 1-6.4.4.2: Device not responding to setup address.
[82150.759129] usb 1-6.4.4.2: device not accepting address 46, error -71
[82150.759415] usb 1-6.4.4-port2: unable to enumerate USB device
```

Am I missing something, or is there a bug here ?

NB: On the same setup, with https://github.com/stm32-rs/stm32f1xx-hal and `cargo embed --features stm32f103 --example usb_serial --release --chip STM32F103C8`, I correctly get a serial tty: 

```
[83377.084346] usb 1-6.4.4.2: new full-speed USB device number 51 using xhci_hcd
[83377.200551] usb 1-6.4.4.2: New USB device found, idVendor=16c0, idProduct=27dd, bcdDevice= 0.10
[83377.200568] usb 1-6.4.4.2: New USB device strings: Mfr=1, Product=2, SerialNumber=3
[83377.200575] usb 1-6.4.4.2: Product: Serial port
[83377.200581] usb 1-6.4.4.2: Manufacturer: Fake company
[83377.200586] usb 1-6.4.4.2: SerialNumber: TEST
[83377.209085] cdc_acm 1-6.4.4.2:1.0: ttyACM0: USB ACM device

Dirbaio, 2023-11-06:
can you post full trace-level logs? `DEFMT_LOG=trace cargo run --bin usb_serial --release`

nim65s, 2023-11-06:
Yes:
```
$ DEFMT_LOG=trace cargo run --bin usb_serial --release
    Finished release [optimized + debuginfo] target(s) in 0.13s
     Running `probe-rs run --chip STM32F103C8 target/thumbv7m-none-eabi/release/usb_serial`
     Erasing sectors ✔ [00:00:02] [########################################################] 37.00 KiB/37.00 KiB @ 14.96 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:55] [##################################################################################################################################################################################################] 37.00 KiB/37.00 KiB @    681 B/s (eta 0s )    Finished in 58.092s
0.000000 TRACE BDCR configured: 00000200
└─ embassy_stm32::rcc::bd::{impl#2}::init @ /home/nim/local/tmp/embassy/embassy-stm32/src/fmt.rs:117 
0.000000 DEBUG rcc: Clocks { sys: Hertz(48000000), pclk1: Hertz(24000000), pclk1_tim: Hertz(48000000), pclk2: Hertz(48000000), pclk2_tim: Hertz(48000000), hclk1: Hertz(48000000), adc: Some(Hertz(6000000)), rtc: Some(Hertz(40000)) }
└─ embassy_stm32::rcc::set_freqs @ /home/nim/local/tmp/embassy/embassy-stm32/src/fmt.rs:130 
0.000000 INFO  Hello World!
└─ usb_serial::____embassy_main_task::{async_fn#0} @ src/bin/usb_serial.rs:30

nim65s, 2023-11-06:
Adding a bit more `info!()`, I can see it is blocking on the `let driver = Driver ::new(p.USB, Irqs, p.PA12, p.PA11);` line

nim65s, 2023-11-06:
And with more `info!()` in `embassy-stm32/src/usb/usb.rs`, it is blocking on `cortex_m::asm::delay(unsafe { crate::rcc::get_freqs() }.sys.0 / 10);`

Dirbaio, 2023-11-06:
I have [this](https://stm32-base.org/boards/STM32F103C8T6-WeAct-Blue-Pill-Plus-Clone) bluepill btw. Maybe you have the other version? Both should work though...

nim65s, 2023-11-06:
Thanks for trying !

My bluepill has micro-usb, and if I disconnect 3.3v from the stlink "VAPP", I get:

```
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An error with the usage of the probe occurred
    1: An error specific to a probe type occurred
    2: Command failed with status JtagGetIdcodeError
```

About this VAPP connection, datasheet says:
> The power supply from the application board is connected to the ST-LINK/V2 debugging and programming board to ensure signal compatibility between the boards.

If I disconnect micro-usb, and only power the bluepill from stlink (with a 3-way cable between bluepill 3v3 and stlink VAPP and VDD), I have the same result (blocking on line `cortex_m::asm::delay(unsafe { crate::rcc::get_freqs() }.sys.0 / 10);`)

This might be my board. I have [that](https://www.az-delivery.de/fr/products/stm32f103c8t6) one. They say it is based on STM32F103C8T6, but I can read on the chip that it is [CH32F103C8T6](https://www.wch-ic.com/products/CH32F103.html).

Dirbaio, 2023-11-06:
> CH32F103C8T6

the CH32F103 is a chip made by a different company aiming to be register-compatible with the stm32f103. It's known to have slight differences though, which might explain why it doesn't work for you.

Unfortunately Embassy only supports the STM32 chips, we don't actively test others. It's still nice to support other chips, if you figure out what the issue is and send a PR I'll merge it.

nim65s, 2023-11-06:
Ok, I understand. I did try a few workarounds on this `cortex_m::asm::delay(unsafe { crate::rcc::get_freqs() }.sys.0 / 10);` line, but I don't get anything working. I guess I'll close this, and try again on a real STM32 :)

closed 2023-11-06 by nim65s