embassy-rs/embassy #2166: SPI1 and USB peripherals conflict in stm32g0b1

cbruiz, 2023-11-09:
Hello.

First of all thank you for this huge contribution.
I'm starting a project leveraging embassy and found an issue with SPI1 and USB in STM32G0B1RETx MCU:
_When I enable SPI1, **even tough not using it**, USB driver gets internally "stall" forever_
However:

- If I do **not** enable SPI1, USB works properly,
- SPI1 works properly despite USB is enabled and not working (stalled).
- I din't experiment the issue with other peripherals.
- I had to enable manually HSI48 and setup Usbsel as other few cases I saw due the RCC initialization in which you are actively progressing.

In order to clearly reproduce and investigate, I just forked and created this branch with your traditional USB example: https://github.com/cbruiz/embassy/tree/stm32g0-usb_spi-rev

This example has a dummy feature: "enable-spi". When enabled (even though not using SPI1 at all, the issue is always reproduced).

Unfortunately, I run out of ideas about how to deal with it. I wouldn't mind to test or try anything you can suggest. Or maybe it's me that I made a mistake in the RCC config...? (Not an expert, sorry). Thank you in advance.

## Demo

### With SPI1 <disabled> -> Fine

I execute the stuff in the device and USB works perfectly.
Log in trace level:
```
/Users/cbruiz/.cargo/bin/cargo run --color=always --package embassy-stm32g0b1-examples --bin usb_serial_plus_spi
    Finished dev [unoptimized + debuginfo] target(s) in 0.12s
     Running `probe-rs run --chip STM32G0B1RETx target/thumbv6m-none-eabi/debug/usb_serial_plus_spi`
     Erasing sectors ✔ [00:00:02] [################################################################################] 128.00 KiB/128.00 KiB @ 48.16 KiB/s (eta 0s )
 Programming pages   ✔ [00:00:04] [################################################################################] 127.00 KiB/127.00 KiB @ 26.62 KiB/s (eta 0s )    Finished in 7.454s
0.000000 TRACE BDCR ok: 00008200
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.000000 DEBUG rcc: Clocks { sys: Hertz(64000000), pclk1: Hertz(64000000), pclk1_tim: Hertz(64000000), hclk1: Hertz(64000000), rtc: Some(Hertz(32000)) }
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:130
0.000030 INFO Hello World!
└─ src/bin/usb_serial_plus_spi.rs:55
0.150451 TRACE allocating type=Interrupt mps=8 interval_ms=255, dir=In
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.150939 TRACE   index=1 addr=64 len=8
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.151397 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=Out
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.151855 TRACE   len_bits = 8400
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.152130 TRACE   index=2 addr=72 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.152557 TRACE allocating type=Bulk mps=64 interval_ms=0, dir=In
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.153015 TRACE   index=2 addr=136 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.153503 INFO USB: device_descriptor used: 18
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.153778 INFO USB: config_descriptor used: 70
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154052 INFO USB: bos_descriptor used: 12
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154327 INFO USB: msos_descriptor used: 0
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154571 INFO USB: control_buf size: 64
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154876 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.155303 TRACE   len_bits = 8400
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.155548 TRACE   index=0 addr=200 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.155975 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.156402 TRACE   index=0 addr=264 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
[... 301 lines cut ...]
0.154174 INFO USB: bos_descriptor used: 12
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154449 INFO USB: msos_descriptor used: 0
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154693 INFO USB: control_buf size: 64
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:143
0.154998 TRACE allocating type=Control mps=64 interval_ms=0, dir=Out
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.155426 TRACE   len_bits = 8400
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.155670 TRACE   index=0 addr=200 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.156066 TRACE allocating type=Control mps=64 interval_ms=0, dir=In
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.156494 TRACE   index=0 addr=264 len=64
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.156921 TRACE enabled
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.157287 TRACE usb: power detected
└─ /Users/cbruiz/Proyectos/embassy/embassy-usb/src/fmt.rs:117
0.157501 TRACE SETUP read waiting
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
0.157714 TRACE wait_enabled OUT WAITING
└─ /Users/cbruiz/Proyectos/embassy/embassy-stm32/src/fmt.rs:117
```

Dirbaio, 2023-11-09:
That is very strange. I don't know what's up, but here are some things you could try toat maybe will give us clues:

- HSI48 needs CRS to be enabled to be precise enough to be USB-compliant. Can you try that? (i'm not sure if it's this, in practice hsi48 always seems to work fine even with no CRS)
- Can you try runnning USB from PLLQ instead of HSI48?
- Can you try SPI1 on different pins? perhaps it's a pin conflict.

cbruiz, 2023-11-12:
Updated with findings. No luck so far

cbruiz, 2023-11-21:
I found couple of details:

- For any reason, USB is not working with PCLKQ input clock in this board.
I saw that any of CubeMx, arduino stm32 and some 3 parties firmware using the same board are working with such setup (PLLSourceMux(HSE 8 MHz)->PLLQ->USB)
Fortunately, HSI48 works fine in the board, so I'm progresing by using HSI48->USB as said
- I found that USB stops responding when I activate **SPI1** with SSM=1 and SSI=1, which is needed for the board wiring. Then, SPI1:
 - With SSM=1, SSI=1 -> USB gets stalled
 - With SSM=0 -> USB works but obviusly SPI does not.
 - With SSM=1, SSI=0 -> USB works but even though I tried to **dynamically** set SSI up and down at the begining and end of a transfer (patching embassy_stm32), SPI1 responds with an error (mode fault) at the first write.
 
 Note: My board is an SKR Mini E3 V3. The SPI1 is wired with an SD Card, which is what I'm testing.

cbruiz, 2023-11-27:
Hello.
It seems to be due a pin conflict and fixed by activating USB pins remap (PA12_RMP and PA11_RMP) in SYSCFG.
Findings and fixes at https://github.com/embassy-rs/embassy/pull/2224

cbruiz, 2023-11-27:
Not an issue in embassy at the end. right?

closed 2023-11-27 by cbruiz