embassy-rs/embassy #2176: cyw43: does not respond to IPv6 Neighbor Solicitation requests

sammko, 2023-11-12:
This is because the cyw43 is not configured with the required HW address for the solicited-node ipv6 multicast address. This is reasonably easy to do manually, provided that a method is added for this in cyw43 Control, something along the lines of:
```rust
    pub async fn add_mcast(&mut self, addr: [u8; 6]) {
        let mut buf = [0; 64];
        self.get_iovar("mcast_list", &mut buf).await;
        let mut n = u32::from_le_bytes([buf[0], buf[1], buf[2], buf[3]]) as usize;
        // check if already present
        for i in 0..n {
            let start = 4 + i * 6;
            if buf[start..start + 6] == addr {
                return;
            }
        }
        let start_new = 4 + n * 6;
        buf[start_new..start_new + 6].copy_from_slice(&addr);
        n += 1;
        buf[..4].copy_from_slice(&n.to_le_bytes());

        self.set_iovar_v::<80>("mcast_list", &buf).await
    }
```
and then:
```rust
    let address = link_local_from_hw(netdev.hardware_address());
    let mut mcast_hw = [0x33, 0x33, 0, 0, 0, 0];
    mcast_hw[2..].copy_from_slice(&address.address().solicited_node().0[12..]);

    control.add_mcast(mcast_hw).await;
```
Doing this works for me and the device starts responding to neighbor solicitations.

Ideally this would be done automatically by embassy-net, but I don't believe a suitable abstraction is in place. I'm not sure how that would look, I guess the Device would have a method for this and embassy-net would call it when an ipv6 config is provided?

Dirbaio, 2023-11-12:
oh wow, nice catch. I guess it slipped through because no tmany people are using ipv6 ;( 

It's very annoying that the multicast address depends on our ipv6 address. :( The the easiest solution would be to declare that embassy-net assumes the underlying device is in a "sane" configuration for ipv4/ipv6 including the multicast stuff, so we could just hardcode enabling multicast in cyw43 init, without adding any API. However that's not possible if cyw43 doesn't know our ipv6 addr. 

Maybe it's possible to enable multicast for the entire solicited-node range? at the cost of performance of course.

Otherwise we'd have to add that API yes...

sammko, 2023-11-12:
Adding all solicited-node multicast addresses is not viable, there's 2^24 of them (33:33: prefix with 32 bits of the solicited-node address, those being ff: and 24 bits from the original address).

I think adding the `add_mcast` to the public API of cyw43 is a nice to have either way, though it does only accept multicast addresses (the I/G bit must be 1). Non-multicast addresses seem to be simply ignored, so the rust-level logic should probably catch that and prevent the user from adding useless addresses. And more graceful error handling when the list gets full. I could put together a PR if you think that is a welcome addition.

Doing it automatically would be nice, but it's probably enough to just document it, considering I'm the first person to run into this anyway

Dirbaio, 2023-11-12:
> Adding all solicited-node multicast addresses is not viable, there's 2^24 of them

okay then :( I was hoping it'd be possible to add a range or something...

> I think adding the add_mcast to the public API of cyw43 is a nice to have either way, though it does only accept multicast addresses (the I/G bit must be 1). Non-multicast addresses seem to be [...]

sounds good then! :+1:

closed 2026-09-01 by Dirbaio