embassy-rs/embassy #2190: STM32F429 LTDC instance cannot be accessible through embassy_stm32

ierturk, 2023-11-15:
Hello,

I'm a newbie here. I'm working on` STM32F429 Discovery`, and working on a display driver by using `LTDC/FMC modules`. The `FMC` and `DMA2D` side seems to work. However there is no instance or driver for the module `LTDC` on `STM32F429`.

At first attempt, I tried to import directly the `LTDC `from `stm32_metapack` but ended up with an error that is multiple definition of the interrupt symbol during linking stage. Also I use directly generated metapack from SVD file.

At the second attempt I tried to impoert the required instaneces as follow

```
use embassy_stm32::peripherals::DMA2D;
use embassy_stm32::peripherals::LTDC;
use embassy_stm32::peripherals::RCC;
```

In this case I ended up following error.
```
9 | use embassy_stm32::peripherals::LTDC;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^^ no `LTDC` in `_generated::peripherals`
```

Any suggestion solve the problem will be appreciated.
Thank you.

Dirbaio, 2023-11-15:
The issue is [stm32-data](https://github.com/embassy-rs/stm32-data) only has [LTDC for F7](https://github.com/embassy-rs/stm32-data/blob/main/stm32-data-gen/src/chips.rs#L244). You'd have to add support for F4.

See instructions in the [readme](https://github.com/embassy-rs/stm32-data#adding-support-for-a-new-peripheral). If you're lucky, the registers between f4 anf f7 are the same and you can do it by simply adding an entry to `perimap`. Otherwise you'd have to add a new YAML and clean it up to make it consistent with the F7 one.

ierturk, 2023-11-15:
Thank you for the prompt reply.

ierturk, 2023-11-28:
I'm able to access the LTDC peripheral by the stm32f4xx-hal drivers. We can close the ticket.

closed 2023-11-29 by Dirbaio