embassy-rs/embassy #2194: Using i2c bus with multiple sensors

maxwen, 2023-11-16:
This is more of a general question on usage then an issue so please assign an appropriate label.
Is there a recommended way in embassy to "share" i2c bus  that is used by multiple sensors (in my case eg. a temp sensor gyroscope and rtc clock). I have a setup like this on a raspberry pico. The sensors are read periodically in tasks based on different timing schedules eg every second or once a minute. As long as I only have two its obviously simple to use i2c0 and i2c1 for one sensor each and that works fine.

PS: I have just started to use Rust on the pico coming from micropython
and I still have a hard time sometimes to understand some of the Rust "challenges"
so if I am just missing something obvious please bear with me

Dirbaio, 2023-11-16:
Yes! this -> https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/index.html

closed 2023-11-17 by Dirbaio

G2-Games, 2024-03-07:
Is there a better example of how to do this? I can't seem to find any examples unfortunately.

svena33, 2024-06-03:
@G2-Games I use the following code to share the bus between tasks.

```
    use core::cell::RefCell;
    use embassy_embedded_hal::shared_bus::blocking::i2c::I2cDevice;
    use embassy_sync::blocking_mutex::NoopMutex;
    
    use sen5x_rs::Sen5x;

    let io = IO::new(peripherals.GPIO, peripherals.IO_MUX);
    let sda = io.pins.gpio6;
    let scl = io.pins.gpio7;

    static I2C_BUS: StaticCell<NoopMutex<RefCell<I2C<I2C0, Blocking>>>> = StaticCell::new();
    let i2c = I2C::new(peripherals.I2C0, sda, scl, 100.kHz(), &clocks, None);
    let i2c_bus = NoopMutex::new(RefCell::new(i2c));
    let i2c_bus = I2C_BUS.init(i2c_bus);

    let i2c_dev1 = I2cDevice::new(i2c_bus);
    let i2c_dev2 = I2cDevice::new(i2c_bus);
    
    spawner.must_spawn(sensor_task_1(
        i2c_dev1,
    ));
    
    spawner.must_spawn(sensor_task_2(
        i2c_dev2,
    ));
    
    ...
    // Then I've declared the task as follows:
    #[embassy_executor::task]
    pub async fn sensor_task_1(
       i2c: I2cDevice<'static, NoopRawMutex, I2C<'static, I2C0, Blocking>>,
    ) {
    let mut sen5x = Sen5x::new(i2c, Delay);
    ...
    }
```
Hope it helps!