embassy-rs/embassy #2202: Panic when using poll_once (embassy-futures)

yanshay, 2023-11-19:
I'm using ESP32 no_std with embassy.

I'm trying to poll on an async function, meaning, I want to check if a result is available w/o waiting for the result.
I wrote, so did the following:
```
let touch_future = touch.async_event();

let touch_poll = poll_once(touch_future);
```

When I run this, at the `poll_once` call I get the panic below.

I see in the panic reference to timer. Inside the async function it may be waiting on a future (irq) using `with_timeout` (from `embassy-time`) which is also embassy feature . Maybe that's related to the panic.

- Am I doing something wrong? Or is it a bug? Or maybe not supported on ESP32?

_( as a side note - I could achieve something similar by using select with a timer of 0 but it seems to me this costs more in speed, because it has to set up a time, etc. )_
```
PanicInfo {
    payload: Any { .. },
    message: Some(
        Found waker not created by the Embassy executor. `embassy_time::Timer` only works with the Embassy executor.,
    ),
    location: Location {
        file: "/Users/yanivshaya/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.3.3/src/raw/waker.rs",
        line: 42,
        col: 9,
    },
    can_unwind: true,
}

Backtrace:

0x40121401
0x40121401 - _embassy_time_schedule_wake
    at ??:??
0x40121851
0x40121851 - _ZN75_$LT$embassy_time..timer..Timer$u20$as$u20$core..future..future..Future$GT$4poll17hd8a631027057219fE
    at ??:??
0x400f2b6a
0x400f2b6a - _ZN98_$LT$futures_util..future..select..Select$LT$A$C$B$GT$$u20$as$u20$core..future..future..Future$GT$4poll17he262a36e387db182E
    at ??:??
0x400f61ac
0x400f61ac - _ZN13xpt2046_esp3224XPT2046$LT$SPI$C$IRQ$GT$11async_event28_$u7b$$u7b$closure$u7d$$u7d$17hc3aaa1af1f07c719E.llvm.16976254322227335400
    at ??:??
0x400eed94
0x400eed94 - _ZN16embassy_executor3raw20TaskStorage$LT$F$GT$4poll17h275433813c7836c0E.llvm.5053872814749477446
    at ??:??
0x401212f6
0x401212f6 - _ZN16embassy_executor3raw8Executor4poll17h18b75c7aad325f33E
    at ??:??
0x400ef99c
0x400ef99c - _ZN14esp_hal_common7embassy8executor6xtensa6thread8Executor3run17hb9555bf349f28770E
    at ??:??
0x400f1ee2
0x400f1ee2 - main
    at ??:??
0x4012150d
0x4012150d - Reset
    at ??:??
0x4011f318
0x4011f318 - ESP32Reset
    at ??:??
```

Dirbaio, 2023-11-19:
where is `poll_once` from? My guess is it's creating its own "no-op" waker.

Using the `integrated-timers` timer queue requires all wakers are created by the executor. As an alternative, you can disable feature `integrated-timers` in `embassy-executor` and enable feature `generic-queue` in `embassy-time`, to enable an executor-independent timer queue. That one works with any waker.

(in general using "manual poll" like `poll_once` is not recommended though, you might want to rewrite your code using `select`/`join` if you want to wait for multiple things. It's much less error prone)

yanshay, 2023-11-19:
> where is `poll_once` from? My guess is it's creating its own "no-op" waker.

It's coming from `embassy_futures`. Isn't that aligned with embassy-executor? 
The code there points to https://docs.rs/embassy-futures/latest/src/embassy_futures/block_on.rs.html#36-45

> (in general using "manual poll" like `poll_once` is not recommended though, you might want to rewrite your code using `select`/`join` if you want to wait for multiple things. It's much less error prone)

What I need to accomplish is, for some time, to call in a loop some function from a UI framework I'm using (to drive animation processing) while still process events from my async functions if such arrive. That library isn't async and in those specific situations require calling it repeatedly. 
Any other way to achieve that?

Dirbaio, 2023-11-19:
> It's coming from embassy_futures. Isn't that aligned with embassy-executor?

Ahhh okay. Yes, these are intended to be used as a "poor man's executor" when you *don't* want to use the full embassy-executor. I hadn't thought about this footgun though. Probably needs documenting.

> What I need to accomplish is, for some time, to call in a loop some function from a UI framework I'm using (to drive animation processing) while still process events from my async functions if such arrive. That library isn't async and in those specific situations require calling it repeatedly.

you can make the UI task do

```rust
loop {
   poll_ui();
   embassy_futures::yield_now().await;
}
```

the `yield_now` lets other tasks run, so you can still use standard async/await for stuff in the other tasks, instead of having to interleave manual polling directly.

yanshay, 2023-11-19:
> you can make the UI task do
> 
> ```
> loop {
>    poll_ui();
>    embassy_futures::yield_now().await;
> }
> ```

So, if I get it right, in my case build a new async function from the code you suggest and then select on it together with my async function?

Dirbaio, 2023-11-19:
yes! the `yield_now` lets the other branch of the `select` run, and also other tasks.

yanshay, 2023-11-19:
Thanks! This is way more elegant.

closed 2025-01-10 by Dirbaio