embassy-rs/embassy #2205: How to send & receive on a single UDPSocket from two tasks at the same time? (or maybe even from one task)

yanshay, 2023-11-20:
The app I'm writing needs to listen on UDP packets and at the same time broadcast (to 255.255.255.255) UDP on the same port to communicate with another system. For this to work the socket needs to bind on the local ip address of the board and the specific port for both operations to work properly.

Each operation (receive and broadcast) separately works well with that configuration.

I want to have one embassy task listen on the socket and process incoming packets, while at the same time another embassy task to send out packets on that same socket.

- What's the proper way to do that if at all doable?

My challenge is that the socket needs to be mut and it's in use at all time by the receive, so how can another task use it to send? Is there a way to clone it so to use the underlying socket from two tasks?

I've done this in python, but there asyncio handled udp datagrams in the event_loop as a special case, allowing me to register a message handler and send through it. I wasn't exposed to the underlying mechanism.
I could implement myself such a mechanism, using a task that does both the receiving and sending, and then accepting send requests through a channel and publishing data received on another channel for other tasks to use, but maybe there's a cleaner technique to do that?
Also, even if using a single task when implementing the above mechanism, it would mean that the listen would stop from time to time when send is required because I can't listen on the socket and at the same time call the send in a single task, but maybe that's also a socket limitation anyway, I'm not that familiar with UDP sockets internals.

lulf, 2023-11-21:
@yanshay Only the bind needs to be &mut, the send_to and recv_from only requires a non-mut reference (see https://docs.embassy.dev/embassy-net/git/default/udp/struct.UdpSocket.html#method.recv_from). 

Once the UdpSocket is created, you can store it in a StaticCell and pass the &'static UdpSocket to each task.

yanshay, 2023-11-21:
> @yanshay Only the bind needs to be &mut, the send_to and recv_from only requires a non-mut reference (see [...]

I tried to take this direction but can't figure out how to do it.

I need to bind the socket in one of the tasks, because the wifi connection that precedes it is all taken care of in tasks and I don't want my main to wait for getting connected to create the socket and bind it before it progress to the main app. So if I pass the socket to the task that will deal with binding, I need to pass it as mut and then it can't be shared between two tasks. Isn't that the case? I tried but can't find how to workaround that.

Also, if connection drops, IP may change, then I'd need to rebind, so need to deal with the socket as mut later on. Don't see how I can do that due to rust constraints on socket mutability. But maybe that's where rust wants to protect me from myself and I need to structure my app differently to deal with it and combine all socket related activity into a single function?

enbyted, 2024-01-18:
Why don't you spawn the second task (taking in `&'static`) from the first (taking in `&'static mut`)?

closed 2025-01-10 by Dirbaio