embassy-rs/embassy #2221: Error when running the `rng` example on `stm32l476rg`

dzamlo, 2023-11-26:
When I try to run the `rng` example on my `stm32l476rg` I get the following error:

```
ERROR probe_rs::util::rtt: 
Error reading from RTT: Control block corrupted: write pointer is 1342572544 while buffer size is 1024 for "up" channel 0 (defmt)    
Frame 0: "HardFault handler. Cause: Escalated BusFault (Precise data access error) at location: 0xffffffff." @ 0x0800f552
Frame 1: "External interrupt #495." @ 0xffffffff
Error: CPU halted unexpectedly.
```

I have adapted the chip feature of `embassy-stm32` in `examples/stm32l4/Cargo.toml` and the chip argument to probe-rs in 
`examples/stm32l4/.cargo/config.toml`.

Other examples work without any issue.

dzamlo, 2023-11-26:
I was able to use `git bisect` to identify the commit where it starts to fail: 5c5e6818195b364199e583eb559b0042d392b3e7

Dirbaio, 2023-11-26:
set `config.rcc.clk48_src = Clk48Src::PLL1_Q`. The default is HSI48, which that chip doesn't have.

(the reason that commit breaks it is it now requires specifying `clk48_src` explicitly, vs before it would try to guess it based on the enabled clocks)