embassy-rs/embassy #2234: Invalid value passed to std::core::asm with arch-xtensa feature enabled.

airblast-dev, 2023-11-30:
I am currently trying to create an async application on my ESP32-WROOM (Xtensa). I used the [esp-idf-template](https://github.com/espressif/esp-idf-template) to get things started. I noticed that the application doesnt compile when ["arch-xtensa", "executor-thread", "nightly"] are enabled. The error states nothing is passed. I tried looking up documentation on the libraries related to the issue, but wasn't able to get any progress. 

I have provided an empty example with minimal crates and feature flags enabled. Even a simple empty main function isnt able to compile. 

`main.rs`:
```rust
#![no_std]
#![no_main]

#[embassy_executor::main]
async fn maina() -> () {}
```
`Cargo.toml`:
```toml
[package]
name = "test"
version = "0.1.0"
authors = ["myname <myname@mail.com>"]
edition = "2021"
resolver = "2"
rust-version = "1.74"

[profile.release]
opt-level = "s"
lto = true

[profile.dev]
debug = true    # Symbols are nice and they don't increase the size on Flash
opt-level = "z"

[dependencies]
esp-idf-sys = {version = "0.33.7"}
embassy-executor = { version = "0.3.3", features = ["arch-xtensa", "executor-thread", "nightly"] }

[build-dependencies]
embuild = "0.31.3"
```

Following is the error's and warning's returned by cargo, after running build, or check.
```
   Compiling embassy-executor v0.3.3
error: cannot use value of type `()` for inline assembly
  --> /home/airblast/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.3.3/src/arch/xtensa.rs:68:62
   |
68 |                     core::arch::asm!("rsil {0}, 5", out(reg) token);
   |                                                              ^^^^^
   |
   = note: only integers, floats, SIMD vectors, pointers and function pointers can be used as arguments for inline assembly

error: cannot use value of type `()` for inline assembly
  --> /home/airblast/.cargo/registry/src/index.crates.io-6f17d22bba15001f/embassy-executor-0.3.3/src/arch/xtensa.rs:78:42
   |
78 |                         "rsync", in(reg) token)
   |                                          ^^^^^
   |
   = note: only integers, floats, SIMD vectors, pointers and function pointers can be used as arguments for inline assembly

error: could not compile `embassy-executor` (lib) due to 2 previous errors
```

airblast-dev, 2023-12-02:
I found [this](https://github.com/embassy-rs/embassy/commit/c211d51c0667b19dc0e70473da4a2742d7405b59) note from @bugadani. 

I used the main macro from `esp-hal-common` with `esp32-hal = { version = "0.16.0", features = ["embassy-executor-thread", "embassy-time", "async", "embassy-time-timg0", "embassy"] }`, and removed the `arch-xtensa` feature flag from embassy-executor. 

This passes a cargo check, however during a build it gives an ldproxy error.

That being said I think this is starting to get out of scope for the issue. Still going to leave this here in case it might point someone in the correct direction or might help finding the issue. For now I will just use threads + block_on for async since I dont see much of an alternative. 

```
error: linking with `ldproxy` failed: exit status: 1
  |
  = note: LC_ALL="C" PATH="/home/airblast/.rustup/toolchains/esp/lib/rustlib/x86_64-unknown-linux-gnu/bin:/home/airbl
ast/.rustup/toolchains/esp/xtensa-esp-elf/esp-13.2.0_20230928/xtensa-esp-elf/bin:/home/airblast/.local/share/nvim/mas
on/bin:/home/airblast/.cargo/bin:/usr/local/bin:/usr/bin:/bin:/usr/local/sbin:~/.local/bin:/opt/android-sdk/platform-
tools:/opt/cuda/bin:/opt/cuda/nsight_compute:/opt/cuda/nsight_systems/bin:/usr/lib/jvm/default/bin:/opt/rocm-5.7.1/bi
n:/opt/rocm-5.7.1/hip/bin:/usr/bin/site_perl:/usr/bin/vendor_perl:/usr/bin/core_perl:/usr/lib/rustup/bin" VSLANG="103
3" "ldproxy" "/tmp/rustcHF7hor/symbols.o" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/d
ebug/deps/uhiffy-e2a1408bf7315cc3.2vp0fc3u9vlus3vg.rcgu.o" "/home/airblast/Documents/test/esp32-smartbulb/target/xten
sa-esp32-espidf/debug/deps/uhiffy-e2a1408bf7315cc3.32ygcyf3encddetd.rcgu.o" "/home/airblast/Documents/test/esp32-smar
tbulb/target/xtensa-esp32-espidf/debug/deps/uhiffy-e2a1408bf7315cc3.3cbmqvvf0pxo2qqm.rcgu.o" "/home/airblast/Document
s/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/uhiffy-e2a1408bf7315cc3.3qmbnq8ygstdquah.rcgu.o" "/home/
airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/uhiffy-e2a1408bf7315cc3.4cunwbmuh2pwyhh
d.rcgu.o" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/uhiffy-e2a1408bf7315cc
3.4oalg1nzdiygj2om.rcgu.o" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/uhiff
y-e2a1408bf7315cc3.rm7urxsnwxmzpww.rcgu.o" "-Wl,--as-needed" "-L" "/home/airblast/Documents/test/esp32-smartbulb/targ
et/xtensa-esp32-espidf/debug/deps" "-L" "/home/airblast/Documents/test/esp32-smartbulb/target/debug/deps" "-L" "/home
/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/build/esp-hal-common-dceaefd49178c2c6/out" 
"-L" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/build/esp32-6fea1dcccfe1079b/out
" "-L" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/build/xtensa-lx-2bbf5a6b90f086
b6/out" "-L" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/build/xtensa-lx-rt-3f2a2
e5d02e58da1/out" "-L" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/build/esp32-hal
-86d887db723b3577/out" "-L" "/home/airblast/.rustup/toolchains/esp/lib/rustlib/xtensa-esp32-espidf/lib" "-Wl,-Bstatic
" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libesp_idf_hal-d114b07232ef880
e.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libatomic_waker-8dbe6fb0
a2884e63.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/liblog-64d8706bf3
c66915.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libnum_enum-7a5b041
70b884518.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libenumset-9faed
209213f2701.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libesp_idf_sys
-50f73e69a646bbd8.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/liblibc-
293f1caebe04a0f3.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libconst_
format-8efa4b5af7656a5c.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/li
bstd-54d7c46ad30858cb.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/deps/libp
anic_abort-4eb518719fb0e1e1.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/dep
s/libobject-354d6c66dcba8d76.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/de
ps/libmemchr-ca566fc8f945d4e9.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/debug/d
eps/libaddr2line-02f09671e283b3a3.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/deb
ug/deps/libgimli-9d9c678d271282aa.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-espidf/deb
ug/deps/librustc_demangle-1185e5de21f34154.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-esp32-e
spidf/debug/deps/libstd_detect-e892c8bfeaa4605e.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtensa-es
p32-espidf/debug/deps/libhashbrown-f2142315ebcdd3fe.rlib" "/home/airblast/Documents/test/esp32-smartbulb/target/xtens
[... cut ...]
            16: _start
          
  = note: some `extern` functions couldn't be found; some native libraries may need to be installed or have their pat
h specified
  = note: use the `-l` flag to specify native libraries to link
  = note: use the `cargo:rustc-link-lib` directive to specify the native libraries to link with Cargo (see https://do
c.rust-lang.org/cargo/reference/build-scripts.html#cargorustc-link-libkindname)

error: could not compile `uhiffy` (bin "uhiffy") due to previous error
```

MathyV, 2023-12-04:
Ah, I didn't realize that esp32-hal progressed that far already, which is awesome of course.

@airblast-dev (and others who come across this in the future) fyi, I got it working by following this example: https://github.com/esp-rs/esp-hal/blob/main/esp32s3-hal/examples/embassy_hello_world.rs

commit ca3891bb (2023-12-03): executor: remove arch-xtensa.

It's been broken for months and nobody has noticed. The `esp*-hal` crates have
much better support.

Fixes #2234
Closes  #1912
PR #2244 (2023-12-03): executor: remove arch-xtensa.
It's been broken for months and nobody has noticed. The `esp*-hal` crates have
much better support.

Fixes #2234
Closes  #1912