embassy-rs/embassy #2239: CYW43 Connection Timeout on Auth Event - RP2040

SteveCookTU, 2023-12-01:
Following the example for `examples/rp/wifi_tcp_server`, the cyw43 controller fails to connection with a timeout status for the auth event.

This was working as intended about a week ago and I have not personally changed anything on my system or network since then.

Log (truncated to around the error):
```
4.326936 DEBUG set ampdu_ba_wsize = [08, 00, 00, 00]
└─ cyw43::control::{impl#0}::set_iovar_v::{async_fn#0}
4.342957 DEBUG set bsscfg:sup_wpa = [00, 00, 00, 00, 01, 00, 00, 00]
└─ cyw43::control::{impl#0}::set_iovar_v::{async_fn#0} 
4.351464 DEBUG set bsscfg:sup_wpa2_eapver = [00, 00, 00, 00, ff, ff, ff, ff]
└─ cyw43::control::{impl#0}::set_iovar_v::{async_fn#0} 
4.360080 DEBUG set bsscfg:sup_wpa_tmo = [00, 00, 00, 00, c4, 09, 00, 00]
└─ cyw43::control::{impl#0}::set_iovar_v::{async_fn#0} 
7.801236 DEBUG === EVENT AUTH: EventMessage { version: 2, flags: 0, event_type: 3, status: 2, reason: 0, auth_type: 0, datalen: 0, addr: [d4, 6c, 6d, 4a, 5d, bf], ifname: [77, 6c, 30, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], ifidx: 0, bsscfgidx: 0, }  []
└─ cyw43::runner::{impl#1}::rx 
7.804450 DEBUG === EVENT SET_SSID: EventMessage { version: 2, flags: 0, event_type: 0, status: 1, reason: 0, auth_type: 0, datalen: 6, addr: [77, 6c, 30, 0, 0, 0], ifname: [77, 6c, 30, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0, 0], ifidx: 0, bsscfgidx: 0, }  [57, 69, 6e, 67, 65, 72]
└─ cyw43::runner::{impl#1}::rx 
7.809246 WARN  JOIN failed with status=1 auth=2
└─ cyw43::control::{impl#0}::wait_for_join::{async_fn#0}
```

The network is listed when scanning on the device as well.

rustc: nightly
OS: Windows 11

Dirbaio, 2023-12-01:
> This was working as intended about a week ago and I have not personally changed anything on my system or network since then.

You mean it worked 1 week ago and it stopped working when you updated Embassy? If that's the case, it would be very helpful if you could bisect it find which commit made it stop working.

SteveCookTU, 2023-12-01:
That's what I'm attempting now, but it'll take a couple hours. If I find any information I'll reply and update.

SteveCookTU, 2023-12-01:
It seems there was an outage in the middle of the night that put the router in an invalid state... I wasn't awake so had no clue it happened and it was preventing new devices from connecting. After a proper reset, everything works as intended.

closed 2023-12-01 by SteveCookTU