embassy-rs/embassy #2256: Support issue for target "riscv32imc-unknown-none-elf"

haijd, 2023-12-06:
When I was developing for my ESP-C3 chip, I used the "embassy-executor" crate for task scheduling. I found that it uses "portable-atomic" for atomic support starting from version "0.3.3". Starting from this version, my project gets the following error when compiling:

```
error[E0433]: failed to resolve: could not find `AtomicU8` in `imp`
   --> /Users/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:659:14
    |
659 |         imp::AtomicU8::is_lock_free()
    |              ^^^^^^^^ could not find `AtomicU8` in `imp`
    |
help: a struct with a similar name exists
    |
659 |         imp::AtomicU128::is_lock_free()
    |              ~~~~~~~~~~
help: consider importing this struct
    |
472 + use core::sync::atomic::AtomicU8;
    |
help: if you import `AtomicU8`, refer to it directly
    |
659 -         imp::AtomicU8::is_lock_free()
659 +         AtomicU8::is_lock_free()
    |

error[E0433]: failed to resolve: could not find `AtomicU8` in `imp`
   --> /Users/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:681:14
    |
681 |         imp::AtomicU8::is_always_lock_free()
    |              ^^^^^^^^ could not find `AtomicU8` in `imp`
    |
help: a struct with a similar name exists
    |
681 |         imp::AtomicU128::is_always_lock_free()
    |              ~~~~~~~~~~
help: consider importing this struct
    |
472 + use core::sync::atomic::AtomicU8;
    |
help: if you import `AtomicU8`, refer to it directly
    |
681 -         imp::AtomicU8::is_always_lock_free()
681 +         AtomicU8::is_always_lock_free()
    |

error[E0412]: cannot find type `AtomicU8` in module `imp`
    --> /Users/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:1384:37
     |
1384 |     fn as_atomic_u8(&self) -> &imp::AtomicU8 {
     |                                     ^^^^^^^^
     |
    ::: /Users/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/imp/fallback/mod.rs:111:9
     |
111  |         pub(crate) struct $atomic_type {
     |         ------------------------------ similarly named struct `AtomicU128` defined here
     |
help: a struct with a similar name exists
     |
1384 |     fn as_atomic_u8(&self) -> &imp::AtomicU128 {
     |                                     ~~~~~~~~~~
help: consider importing this struct
     |
472  + use core::sync::atomic::AtomicU8;
     |
help: if you import `AtomicU8`, refer to it directly
     |
1384 -     fn as_atomic_u8(&self) -> &imp::AtomicU8 {
1384 +     fn as_atomic_u8(&self) -> &AtomicU8 {
     |

error[E0412]: cannot find type `AtomicU8` in module `imp`
    --> /Users/root/.cargo/registry/src/index.crates.io-6f17d22bba15001f/portable-atomic-1.5.1/src/lib.rs:1387:56
[... 757 lines cut ...]
  "-C", "link-arg=-Tlinkall.x",

  # Required to obtain backtraces (e.g. when using the "esp-backtrace" crate.)
  # NOTE: May negatively impact performance of produced code
  "-C", "force-frame-pointers",

  # comment the cfgs below if you do _not_ wish to emulate atomics.
  # enable the atomic codegen option for RISCV
  "-C", "target-feature=+a",
  # tell the core library have atomics even though it's not specified in the target definition
  "--cfg", "target_has_atomic_load_store",
  "--cfg", 'target_has_atomic_load_store="8"',
  "--cfg", 'target_has_atomic_load_store="16"',
  "--cfg", 'target_has_atomic_load_store="32"',
  "--cfg", 'target_has_atomic_load_store="ptr"',
  # enable cas
  "--cfg", "target_has_atomic",
  "--cfg", 'target_has_atomic="8"',
  "--cfg", 'target_has_atomic="16"',
  "--cfg", 'target_has_atomic="32"',
  "--cfg", 'target_has_atomic="ptr"',
]
```

The compilation tool chain of the "nightly" channel is used.

taiki-e, 2023-12-06:
See https://github.com/embassy-rs/static-cell/issues/10#issuecomment-1842037543.

Dirbaio, 2023-12-08:
can you send a PR?

closed 2023-12-08 by Dirbaio