embassy-rs/embassy #2259: STM32H7 SpiDeviceWithConfig does not properly apply the config

SteveNguyen, 2023-12-06:
I am trying to use SpiDeviceWithConfig https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/asynch/spi/struct.SpiDeviceWithConfig.html
in order to share a spi bus between different device drivers with different spi config (on a STM32H7).
I am using a shared_bus as explained here: https://docs.embassy.dev/embassy-embedded-hal/git/default/shared_bus/blocking/spi/index.html and it seems to work ok.
My problem is that it seems that the Config is not correctly applied for each of my device.
After a bit of debugging I can confirm that the correct set_config() is called: https://github.com/embassy-rs/embassy/blob/83bed7bca476a71da6cfbdb074fdd59862929750/embassy-stm32/src/spi/mod.rs#L325
with the correct config but if I check the actual config of the spi after this call, it is stuck at the spi init state.
After a bit of tinkering, it happens that if I do in `set_config()` as it is done in `new_inner()`: https://github.com/embassy-rs/embassy/blob/83bed7bca476a71da6cfbdb074fdd59862929750/embassy-stm32/src/spi/mod.rs#L213
with the `T::enable_and_reset()` and setting the registers, everything seems to work as intended, i.e. the config is correctly set for each of my device at each spi call.

To clarify a little bit my code looks something like that:

`   let mut spi_config = spi::Config::default();
    spi_config.frequency = embassy_stm32::time::Hertz(1_000_000);
    spi_config.bit_order = spi::BitOrder::MsbFirst;
    spi_config.mode = spi::MODE_1;

    let spi = spi::Spi::new(
        peri,
        sck,
        mosi,
        miso,
        NoDma,
        NoDma,
        spi_config,
    );
    let spi_bus: Mutex<NoopRawMutex, _> = Mutex::new(RefCell::new(spi));

    let mut spi_device1_config = spi::Config::default();
    spi_device1_config.frequency = embassy_stm32::time::Hertz(1_000_000);
    spi_device1_config.mode = spi::MODE_3;
    spi_device1_config.bit_order = spi::BitOrder::MsbFirst;

    let mut spi_device2_config = spi::Config::default();
    spi_device2_config.mode = spi::MODE_3;
    spi_device2_config.frequency = embassy_stm32::time::Hertz(1_000_000);
    spi_device2_config.bit_order = spi::BitOrder::MsbFirst;

    let spi_device1 = SpiDeviceWithConfig::new(
        &spi_bus,
        Output::new(device1_cs, Level::High, Speed::Medium),
        spi_device1_config,
    );

    let spi_device2 = SpiDeviceWithConfig::new(
        &spi_bus,
        Output::new(device2_cs, Level::High, Speed::Medium),
        spi_device2_config,
    );

    loop{
        spi_device1
            .transfer_in_place(&mut transfer_data)
            .map_err(|e| {
                error!("!!! Error SPI {:?}!!!", e);
                embassy_stm32::spi::Error::Framing
            })?;
        Timer::after(Duration::from_millis(1)).await;
        spi_device2
            .transfer_in_place(&mut transfer_data)
            .map_err(|e| {
                error!("!!! Error SPI {:?}!!!", e);
                embassy_stm32::spi::Error::Framing
            })?;

        Timer::after(Duration::from_millis(1)).await;
    }
`

By adding some debug I clearly see the `set_config()` being called for each transfer but the bus config keeps stuck at the `spi_config` state.

Did I miss something?

Dirbaio, 2023-12-08:
Nice find! :eyes: 

it could be taht SPI needs to be disabled (SPE=false) before changing the config? This might be why an RCC reset fixes it, because it disables it. We should avoid a full RCC reset, it's too heavyweight.

If you figure it out, could you send a PR fixing this?

SteveNguyen, 2023-12-14:
Well, looks like you are right :) disabling/enabling the SPI at each reconfigure seems to do the trick.
I inserted a:
```
T::REGS.cr1().modify(|w| {
  w.set_spe(false);
});

...

T::REGS.cr1().modify(|w| {
  w.set_spe(true);
});
```
In this block: https://github.com/embassy-rs/embassy/blob/83bed7bca476a71da6cfbdb074fdd59862929750/embassy-stm32/src/spi/mod.rs#L344C10-L344C10
which corresponds to my chip. I can't say for the other cfg but it is probably the same deal?

sebgab, 2025-04-22:
I have the same problem with an STM32U5, resetting disabling and then re-enabling the SPI when reconfiguring as suggested does work.

I've created a PR with my fix #4121.

PR #4121 (2025-04-22): Fix STM32H7 SpiDeviceWithConfig does not properly apply the config
The new SPI configuration applied when using SpiDeviceWithConfig did previously not get applied as the SPI had to be disabled and re-enabled for it to have an effect. This is implemented by this PR and should resolve the issue.

It has been tested and is working on an STM32U5 which displayed the same problem that the STM32H7 has as described in #2259.

Closes #2259
commit d8631f96 (2025-04-22): Disable SPI before changing config, then re-enable SPI on STM32

Fixes #2259
closed 2025-04-22 by Dirbaio