embassy-rs/embassy #2274: RPi Pico W with Picoprobe doesn't connect after the first time

m93a, 2023-12-10:
I'm trying to run the example programs on my Raspberry Pi Pico W using a Picoprobe setup, as described in the RPi Pico [manual](https://datasheets.raspberrypi.com/pico/getting-started-with-pico.pdf).  I ran the following commands:
```
$ git clone https://github.com/embassy-rs/embassy.git
$ cd embassy/examples/rp
$ cargo build --bin blinky --release
$ cargo run --bin blinky --release
```
It worked insofar that I got the "led on" and "led off" logs in my terminal. However the onboard LED did not blink because I mistakenly ran `blinky` and not `wifi_blinky`. I immediately tried to run the correct program, however I wasn't so successful:
```
$ cargo build --bin wifi_blinky --release
$ cargo run --bin wifi_blinky --release

    Finished release [optimized + debuginfo] target(s) in 0.07s
     Running `probe-rs run --chip RP2040 target/thumbv6m-none-eabi/release/wifi_blinky`
Error: Connecting to the chip was unsuccessful.

Caused by:
    0: An ARM specific error occurred.
    1: An error occurred in the communication with an access port or debug port.
    2: Target device did not respond to request.
```

In fact, I wasn't able to recreate the initial success, as from then on I only get this error. I am not the only one to face this error, as evidenced by [this thread on Rust forums](https://users.rust-lang.org/t/probe-rs-fails-to-work-after-first-time-use-successful/103234/4). However, the developer over there used a different probe which had a NRST pin – I don't know whether Picoprobe supports the reset pin functionality, or how to set it up. Therefore, I have been unsuccessful in implementing the solution described there.

Another issue which might be related is #757, but I haven't been able to use pico-debug either (and the project has been archived since).

finnkauski, 2023-12-23:
So my experience trying it out this evening is also quite disapointing. Looks similar to yours yet I don't even get the luxury of the connection error it seems. 

I have the 2 pico setup. I run blinky fine, but when I try running the wifi_blinky it crashes saying the following: 

```
<snip>
0.494460 DEBUG shared_addr 000177ac
└─ cyw43::runner::{impl#1}::log_init::{async_fn#0} @ /Users/definitely.artful/projects/pico/embassy/cyw43/src/fmt.rs:130
0.494658 DEBUG wifi init done
└─ cyw43::runner::{impl#1}::init::{async_fn#0} @ /Users/definitely.artful/projects/pico/embassy/cyw43/src/fmt.rs:130
0.494711 ERROR panicked at 'embassy-executor: task arena is full. You must increase the arena size, see the documentation for details: https://docs.embassy.dev/embassy-executor/'
└─ embassy_executor::_export::{impl#2}::alloc @ /Users/definitely.artful/projects/pico/embassy/embassy-executor/src/fmt.rs:106
0.494742 ERROR panicked at /Users/definitely.artful/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.5/src/lib.rs:368:5:
explicit panic
└─ panic_probe::print_defmt::print @ /Users/definitely.artful/.cargo/registry/src/index.crates.io-6f17d22bba15001f/panic-probe-0.3.1/src/lib.rs:104
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::debug_info: UNWIND: Callsite for inlined function None
DEBUG probe_rs::debug::unit_info: Found DIE, now checking for inlined functions: name=None
Frame 0: HardFault @ 0x1000bbb6
Frame 1: __udf @ 0x1000759e inline
       /Users/definitely.artful/.cargo/registry/src/index.crates.io-6f17d22bba15001f/cortex-m-0.7.7/src/../asm/inline.rs:181:5
Frame 2: udf @ 0x000000001000759e inline
<snip>
``` 
I run blinky fine multiple times _until_ I try running the wifi one and it crashes with this error. After that I can't run anything else on my device until I open it up with BOOTSEL and load up some random uf2 into it. This seems to reset it enough for me to try running stuff again. 

I have tried adding in my `.cargo/config`  the following env value: 
`EMBASSY_EXECUTOR_TASK_ARENA_SIZE="100"` 
and others like 4, 10 - no dice.

m93a, 2024-01-01:
I tried to create the whole setup again, this time using another Pico **W** as the probe, and this time everything works as expected. Each time I plug the probe into the PC, I have to hold BOOTSEL on the programmed pico for a sufficiently long time – maybe the problem was that I failed to do that before.

I don't have to set the task arena size for the examples to work – so far I've tried `wifi_blinky`, `wifi_scan`, `usb_ethernet` and `usb_hid_keyboard`.

I'm using the [official picoprobe](https://github.com/raspberrypi/picoprobe/releases/tag/picoprobe-cmsis-v1.0.3). I also tried [yapicoprobe](https://github.com/rgrr/yapicoprobe) which includes support for a _reset_ pin, hoping that I wouldn't need to press BOOTSEL each time I upload a program, but the upload process was quite unstable and usually failed, so I went back to the official.

I guess I'll close this issue as both me and @finnkauski have solved our issues (or at least found walkarounds), but I'll be happy to reopen if you think this needs more action.

closed 2024-01-01 by m93a