embassy-rs/embassy #2278: [embassy-stm32]: SPIv1 is not functional in `rx_only` mode

jamesmunns, 2023-12-12:
SPIv1 currently allows users to create `new_rxonly` (as well as `new_txonly`, but I have not tested that), however it still requires a "real" DMA instance for `RxDma`.

I did this, which compiled, and gave me a `Spi` that implemented `embedded-hal-async::SpiBus`.

However, when you actually *use* the `Spi` instance, all SpiBus methods, even `read`, will hang indefinitely.

I believe in the `new` function, where RXONLY mode is set:

```rust
w.set_rxonly(vals::Rxonly::OUTPUTDISABLED);
```

Must have the side effect of also disabling the DMA machinery, which means that when we later do real/dummy writes, the DMA transaction never completes, leading the function to block forever.

I even tried replacing all operations with `read` in my `SpiDevice` adapter (which manages CSn pins):

```rust
for op in operations {
    info!("Doing operation...");
    match op {
        Operation::Read(r) => spi.read(r).await?,
        // Operation::Write(w) => spi.write(w).await?,
        Operation::Write(_w) => {},
        Operation::Transfer(t1, _t2) => spi.read(t1).await?,
        Operation::TransferInPlace(t) => spi.read(t).await?,
        Operation::DelayNs(d) => {
            Timer::after(Duration::from_nanos((*d).into())).await;
        }
    }
}
```

However this also hung.

From @Dirbaio in chat:

> the DMA generics are a bit WIP in stm32
>
> on other periphs like uart, read requires rxdma, write requires txdma
>
> spi is cursed because read requires dummy write and write requires dummy read
>
> and the DMA generics are kinda WIP because the end goal is to remove them
>
> and do like embassy-rp, where you have Spi<'d, Blocking> (without DMA), Spi<'d, Async> (with DMA), but no need to put the actual DMA channels in the generics
>
> (Spi would maybe also have BlockingTxOnly, BlockingRxOnly, AsyncTxOnly, AsyncRxOnly, or perhaps there's no point if it's still going to impl the whole spi traits)
>
> but doing this requires having AnyChannel which we dont' have now... and it's annoying on H7s that mix DMA and BDMA
>
> 😵‍💫

As a workaround, users will need to map a "dummy" pin to allow for use.

jamesmunns, 2023-12-12:
This was specifically tested on the STM32F405, btw.

Dirbaio, 2024-06-02:
fixed in #3027

closed 2024-06-02 by Dirbaio