embassy-rs/embassy #2279: embassy-boot: Hitting assert in `assert_partitions` when DFU partition size equals ACTIVE partition size

plaes, 2023-12-13:
I feel that asserts defined in the [`assert_partitions`](https://github.com/embassy-rs/embassy/blob/14f41a71b6ea9dedb4ee5b9c741fe10575772c7d/embassy-boot/boot/src/boot_loader.rs#L401) function need some kind of explanations as I can't figure out the need for this one I'm hitting: `assert!(dfu.capacity() as u32 - active.capacity() as u32 >= page_size);`

In the linker configuration I have set following:
```
MEMORY
{
  /* NOTE 1 K = 1 KiBi = 1024 bytes */
  /* Softdevice - s132_nrf52_7.3.0: 152kB (0x26000 bytes including MBR) */
  MBR                               : ORIGIN = 0x00000000, LENGTH = 4K
  SOFTDEVICE                        : ORIGIN = 0x00001000, LENGTH = 148K
  ACTIVE                            : ORIGIN = 0x00026000, LENGTH = 164K 
  DFU                               : ORIGIN = 0x0004f000, LENGTH = 164K
...
```

Dirbaio, 2023-12-13:
DFU needs to be 1 page bigger than ACTIVE. This is needed for the swap algorithm to work.

lulf, 2023-12-13:
@plaes There is also some docs here that might help explaining why:

* https://embassy.dev/book/dev/bootloader.html
* https://blog.drogue.io/firmware-updates-part-1/

plaes, 2023-12-13:
I think we should leave this one still open to document the partition requirements in the top-level library.
For example that we need at least 4 partitions at minimum (FLASH, BOOTLOADER_STATE, ACTIVE and DFU) and how things need to be swapped around between app / bootloader.

lulf, 2023-12-13:
Fixed in #2283

closed 2023-12-13 by lulf