embassy-rs/embassy #2305: embassy-rp UART e_hal_nb recv has duplicate case

jamesmunns, 2023-12-18:
https://github.com/embassy-rs/embassy/blob/88e77c733cc3d52a2bd20e9e25446fe1d37c830b/embassy-rp/src/uart/mod.rs#L820-L839

Here, `dr.fe()` is checked twice, and I think we can never actually yield data?

Not sure if it is supposed to be more like this (from the ehal 0.2 impl):

```rust
impl<'d, T: Instance, M: Mode> embedded_hal_02::serial::Read<u8> for UartRx<'d, T, M> {
    type Error = Error;
    fn read(&mut self) -> Result<u8, nb::Error<Self::Error>> {
        let r = T::regs();
        if r.uartfr().read().rxfe() {
            return Err(nb::Error::WouldBlock);
        }

        let dr = r.uartdr().read();

        if dr.oe() {
            Err(nb::Error::Other(Error::Overrun))
        } else if dr.be() {
            Err(nb::Error::Other(Error::Break))
        } else if dr.pe() {
            Err(nb::Error::Other(Error::Parity))
        } else if dr.fe() {
            Err(nb::Error::Other(Error::Framing))
        } else {
            Ok(dr.data())
        }
    }
}
```

Dirbaio, 2023-12-18:
why are you even using nb :grimacing: 

jk. yeah this looks like a bug, could you send a PR?

jamesmunns, 2023-12-18:
I'm not, but I have R-A set to do `clippy` instead of `check`, and so it lit up a big yellow highlight. I'll see if I can send a PR, I wasn't sure if it was different from ehal 0.2 for some other reason.

PR #2306 (2023-12-18): Fix nb on rp uart
Closes #2305 

I'm not sure how to test this, but it now matches ehal v0.2 impl.