embassy-rs/embassy #2322: embassy_sync::zerocopy_channel isn't MPSC

darkwater, 2023-12-19:
```rust
impl<'a, M: RawMutex, T> Sender<'a, M, T> {
  pub fn borrow(&mut self) -> Sender<'_, M, T>
```

The additional Sender takes the lifetime of the borrow call, making the original Sender unusable until it's dropped. Is this channel type destined to be SPSC, or is this just a mistake in the API? And if it's really MPSC, couldn't it implement Clone and perhaps Copy?

Dirbaio, 2023-12-19:
It's intended to be SPSC, yes. The way "zero-copy" is done makes it hard to do MPSC, you'd have to keep track which "slots" are reserved to which senders, vs with SPSC it's trivial, it's only the "next" empty slot that can ever be reserved.

mdrssv, 2025-01-17:
> It's intended to be SPSC, yes. The way "zero-copy" is done makes it hard to do MPSC, you'd have to keep track which "slots" are reserved to which senders, vs with SPSC it's trivial, it's only the "next" empty slot that can ever be reserved.

Wouldn't it be sufficient to turn `front` & `back` into an array of `[Option<usize>; N]` where `N` represents the max number of sender & receiver instances?