embassy-rs/embassy #2344: Async Radio driver for embassry_nrf

guissalustiano, 2023-12-22:
Hi! I'm Interested in developing some Bluetooth tools for Rust (as my bachelor's thesis),
as a first step, I would like to implement the Radio driver with async support. Do you think it makes sense?

I started the development [here](https://github.com/guissalustiano/embassy/tree/feat/radio) and I'm unsure about the better API approach. The radio hardware varies between the boards, a quick resume:

| Board     | BLE 1M | BLE 2M | BLE 512K | BLE 128K | 802.15.4 | Nordic 1M | Nordic 2M |
|-------------|------------|-------------|---------------|---------------|-------------|----------------|---------------|
| nrf52805 | ✅ | ✅  | ❌ | ❌ | ❌ | ✅ | ✅ |
| nrf52810 | ✅ | ✅  | ❌ | ❌ | ❌ | ✅ | ✅ |
| nrf52811 | ✅ | ✅  | ✅ | ✅ | ✅ | ✅ | ✅ |
| nrf52820 | ✅ | ✅  | ✅ | ✅ | ✅ | ✅ | ✅ |
| nrf52832 | ✅ | ✅  | ❌ | ❌ | ❌ | ✅ | ✅ |
| nrf52833 | ✅ | ✅  | ✅ | ✅ | ✅ | ✅ | ✅ |
| nrf52840 | ✅ | ✅  | ✅ | ✅ | ✅ | ✅ | ✅ |
| nrf5340   | ✅ | ✅  | ✅ | ✅ | ✅ | ✅ | ✅ |
| nrf9160   | ❌ | ❌ | ❌ | ❌ | ❌ | ❌ | ❌ |

## Desing
My first idea was to create a generic radio driver, but I'm uncomfortable with the 802.15.4 and Nordic proprietary protocol,
so now I'm tending to implement the `BleRadio` driver, with the configuration default BLE configuration, and trying to support both BLE 2M/1M and 512K/128K.

## Steps
I would like to break this into multiple small PRs:
- [ ] Implement the Radio peripheral to all boards with blocking transmission for BLE 1M and 2M
- [ ] Implement the async transmission for BLE 1M and 2M
- [ ] Implement the blocking reception for BLE 1M and 2M
- [ ] Implement the async reception for BLE 1M and 2M
- [ ] Implement the transmission (blocking and async) for BLE 512Kb/128Kb
- [ ] Implement the reception (blocking and async) for BLE 512Kb/128Kb

## Future improvements // Thinks that I'm not caring about now
- I'm not planning to use any shortcuts, but this could improve the whole process
- Maybe one buffered version of the Radio makes sense
- Maybe channel support makes sense
- I'm not sure how this could integrate with the AES to encrypt packets on the fly

lulf, 2023-12-26:
Hi @guissalustiano I think it's a great initiative, but also a very big one. It's also a question how much of this needs to be in   embassy or not. For instance, the LoRa and LoRaWAN work was recently moved out of the embassy repo into a separate organization https://github.com/lora-rs, and I'm wondering if it would be better for work like this to keep the HAL-dependent parts in this repo, and the HAL-independent parts of the stack outside as much as possible. This also has other benefits such as allowing other HALs to benefit from your work.

guissalustiano, 2024-01-25:
Hey, just an update.
Based on @lulf feedback I decided to move the work to a separate [repo](https://github.com/guissalustiano/jewel-rs), there are a lot of agnostic functions like the package structures and serialization. And also make it easier to iterate over the interface. I tried to define a basic trait there that I plan to use with the embassy hal to implement other layers.

Here are two working low-level examples using the current trait:
https://github.com/guissalustiano/jewel-rs/blob/dc39b05428062ade6dc3df1adbe7e0d5061ae594/examples/nrf52840/src/bin/beacon.rs#L27-L45

https://github.com/guissalustiano/jewel-rs/blob/5b35cc591206d3c27573ede729a20086c47e329c/examples/nrf52840/src/bin/scanner.rs#L26-L36

guissalustiano, 2024-02-07:
First high-level API for advertising implemented!
https://github.com/jewel-rs/jewel/blob/584cd5497080e38b817d2a5b39ec37764723f9c0/examples/nrf52840/src/bin/beacon.rs#L21-L37

closed 2024-02-28 by lulf