embassy-rs/embassy #2345: `NoPin` missing in nRF HAL

trembel, 2023-12-22:
Trying to use `NoPin` for a smartled using SPI (only MOSI needed), I got the following error:
```
9 | use embassy_nrf::gpio::NoPin;
  |     ^^^^^^^^^^^^^^^^^^^^^^^^ no `NoPin` in `gpio`
```
and apparently it is not available, although being "used" in
https://github.com/embassy-rs/code-size-bench/blob/cbdb1831218afae43af373f99623d7158ecaf5f7/drogue-device/src/main.rs#L10
https://github.com/embassy-rs/code-size-bench/blob/cbdb1831218afae43af373f99623d7158ecaf5f7/async-embassy/src/main.rs#L9
https://github.com/embassy-rs/code-size-bench/blob/cbdb1831218afae43af373f99623d7158ecaf5f7/async-block-on/src/main.rs#L7

so for me, or it is removed in `code-size-bench`, or added to the HAL; Is there a reason for it being removed (as I assume it was in the HAL earlier)?

Dirbaio, 2023-12-22:
the `code-size-bench` repo is super old, don't use it as a reference.

`NoPin` was removed in favor of adding constructor variants to the drivers that take more or less pins.

You can remove MOSI with [`new_txonly`](https://docs.embassy.dev/embassy-nrf/git/nrf52805/spim/struct.Spim.html#method.new_txonly), but it still requires a SCK pin. You'll have to sacrifice an unused pin for SCK, hopefully you have one.

If oyu want to not have SCK either, it'd need a [`new_txonly_nosck`](https://docs.embassy.dev/embassy-stm32/git/stm32h743zi/spi/struct.Spi.html#method.new_txonly_nosck) constructor like there's in `embassy-stm32`. A pull request for this would be very welcome.

swanandx, 2024-01-03:
> You can remove MISO with [new_txonly](https://docs.embassy.dev/embassy-nrf/git/nrf52805/spim/struct.Spim.html#method.new_txonly)
>
but `new_txonly` takes `miso` right? ( or did u mean MOSI here? )

if it is fine, may I open the PR with `new_txonly_nosck()`? ( and necessary changes in `new_inner` & others )

Dirbaio, 2024-01-03:
> may I open the PR with new_txonly_nosck()?

of course, a PR adding this would be very welcome!

Dirbaio, 2024-03-03:
newer HAL design doesn't need NoPin, it does multiple constructors instead. like https://github.com/embassy-rs/embassy/pull/2395

closed 2024-03-03 by Dirbaio