embassy-rs/embassy #2363: Program sometimes hangs in spinlock

uzytkownik, 2023-12-27:
I tried to add a bootloader to my program. When it started hanging I started digging into the issue. Finally it seemed that critical section implementation was responsible:

```
#![no_std]
#![no_main]
#![feature(type_alias_impl_trait)]

use defmt::*;
use embassy_executor::Spawner;
use embassy_rp::gpio;
use embassy_time::{Duration, Timer};
use gpio::{Level, Output};
use {defmt_rtt as _, panic_probe as _};

#[cortex_m_rt::entry]
fn main() -> ! {
    info!("Program start");
    let p = embassy_rp::init(Default::default());
    let mut led = Output::new(p.PIN_25, Level::Low);

    loop {
        info!("led on!");

        info!("led off!");
    }
}
```

Program hanged on first line:

```
#0  rp_pac::common::Reg<u32, rp_pac::common::RW>::read<u32, rp_pac::common::RW> (self=<optimized out>) at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/rp-pac-6.0.0/src/common.rs:42
#1  embassy_rp::critical_section_impl::Spinlock<31>::try_claim<31> () at /home/mpiechotka/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4634316/embassy-rp/src/critical_section_impl.rs:106
#2  embassy_rp::critical_section_impl::RpSpinlockCs::acquire () at /home/mpiechotka/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4634316/embassy-rp/src/critical_section_impl.rs:56
#3  embassy_rp::critical_section_impl::{impl#0}::acquire () at /home/mpiechotka/.cargo/git/checkouts/embassy-9312dcb0ed774b29/4634316/embassy-rp/src/critical_section_impl.rs:28
#4  0x100011fc in embassy_rp::critical_section_impl::_critical_section_1_0_acquire () at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:282
#5  0x100006ae in critical_section::acquire () at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/critical-section-1.1.2/src/lib.rs:183
#6  defmt_rtt::{impl#0}::acquire () at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-rtt-0.4.0/src/lib.rs:54
#7  0x10000704 in defmt_rtt::_defmt_acquire () at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-rtt-0.4.0/src/lib.rs:43
#8  0x1000020e in defmt::export::acquire () at /home/mpiechotka/.cargo/registry/src/index.crates.io-6f17d22bba15001f/defmt-0.3.5/src/export/mod.rs:55
```

The hang wasn't 100% consistent.

uzytkownik, 2023-12-27:
Apparently it is a well known bug of RP2040 based on channel discussion.

Dirbaio, 2023-12-27:
this is #1736, closing in favor of that one

closed 2023-12-27 by Dirbaio