embassy-rs/embassy #2365: [Question] Embedded async relationship with Symmetric Multiprocessing (SMP)?

awsa2ron, 2023-12-28:
It's a open question. 
I found this [Symmetric Multiprocessing (SMP) with FreeRTOS ](https://www.freertos.org/symmetric-multiprocessing-introduction.html)

Just wondering deeper insight form you.

lights0123, 2024-01-02:
SMP is totally supported—one of embassy's supported chips, the RP2040, operates in SMP with dual cores. Here's an example of activating the second core with embassy: https://github.com/embassy-rs/embassy/blob/main/examples/rp/src/bin/multicore.rs

lights0123, 2024-01-02:
I don't believe Embassy has any sort of multicore scheduler though—you spawn tasks on a specific core and they will never switch, so tasks on the same core will never run in parallel. Tasks running on different cores will run in parallel, though.

enbyted, 2024-01-18:
Small clarification on the terms used. I think embassy does not support **SMP** (S = Symmetric), however it supports **AMP** (A = Asymmetric).
The difference being that in SMP any processing unit takes work from a shared queue (as is the case in FreeRTOS SMP), while in AMP each processing unit has a predefined list of tasks that it may execute.

I guess that you could implement something like having a few executors that are shared between cores, that you then `.pool()` whichever executor has work to do from whichever core is free, which would achieve SMP, but it's not build-in, I don't think.

closed 2025-01-10 by Dirbaio