embassy-rs/embassy #2400: embassy-stm32 uart baudrate wrong with higher hse clock

WeissbMa, 2024-01-03:
I tried pushing the baud rate of the usart_dma example, however doubling the clock seems to halve the baud rate.

```
#[embassy_executor::main]
async fn main(_spawner: Spawner) {
    let mut stm32config : embassy_stm32::Config = Default::default();
    // stm32config.rcc.hse=Some(embassy_stm32::time::Hertz(16_000_000)); // <= disable to break baudrate
    let p = embassy_stm32::init(stm32config);

    let mut config = Config::default();
    config.baudrate=1_000_000;
    config.data_bits=DataBits::DataBits8;
    config.stop_bits=StopBits::STOP1;
    config.parity=Parity::ParityNone;
    config.assume_noise_free=false;
    config.invert_tx=false;
    config.invert_rx=false;

    let mut usart = Uart::new(p.USART1, p.PA10, p.PA9, Irqs, p.DMA1_CH4, NoDma, config).unwrap();

    let mut s: String<128> = String::new();
    core::write!(&mut s, "Hallo:)\r\n").unwrap();
    loop {

        unwrap!(usart.write(s.as_bytes()).await);
        Timer::after_millis(100).await;
    }
}
```

Switching the clock from 8MHz to 16MHz at 1MBaud seems to activate the code path in embassy-stm32/src/usart/mod.rs:1063, where over8 remains false.

I've checked the datasheet and the calculation of the brr seems correct.

I also couldn't find anything related to the baudrate in the errata sheet.

Can please someone try to reproduce the issue, or give me some pointers how to debug this properly? I'm running out of ideas.

Dirbaio, 2024-01-03:
HSE *tells* embassy-stm32 the frequency of the HSE in your board, it doesn't *set* it.

If you tell embassy-stm32 your board has a 16Mhz HSE, it'll believe you and calculate baudrate prescalers accordingly. If you lied and the board actually has a 8Mhz HSE, baudrate will be 2x slower.

WeissbMa, 2024-01-03:
This actually makes a lot of sense. I feel very stupid now, but I've been pulling out my hair as I couldn't find anything while debugging.

Thanks a lot :)

closed 2024-01-03 by WeissbMa